stereolabs/zed-sdk #473: ZED Custom Box Example Segmentation fault

lacie-life, 2022-03-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I cloned example tensorrt_yolov5 with ZED custom box input. I also download the yolov5s.wts from this repo. All the build and generate tensorrt engine is fine. But when i run the executable file with engine, it run a short time then it crash.

### Steps to Reproduce

1.  Clone code in https://github.com/stereolabs/zed-examples/tree/master/object%20detection/custom%20detector/cpp/tensorrt_yolov5
2. Download yolov5s.wts file in https://download.stereolabs.com/sample_custom_objects/yolov5s.wts.zip
3. Build and generate tensorrt engine
4. Run

### Expected Result

Run without crash

### Actual Result

[ZED][INFO] Logging level INFO
[ZED][INFO] [Init]  Depth mode: ULTRA
[ZED][INFO] [Init]  Camera successfully opened.
[ZED][INFO] [Init]  Sensors FW version: 776
[ZED][INFO] [Init]  Camera FW version: 1523
[ZED][INFO] [Init]  Video mode: HD1080@30
[ZED][WARNING] Unhandled internal exception (slutils)
Segmentation fault (core dumped)

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18.04(Jetpack 4.6)
Device: Jetson AGX Xavier
ZED SDK: ZED_SDK_Tegra_JP46_v3.7.0
Model: Yolov5s
```

lacie-life, 2022-03-30:
I have tested with ZED SDK 3.6 and everything worked fine. I think have some problems with the new release.

closed 2022-03-30 by lacie-life

arremebg, 2023-11-14:
We are having the same error now.... How did you fixed? Does it need to install again the library?

lacie-life, 2023-11-14:
I just downgraded the ZED SDK to 3.6 and this error disappeared in the newest ZED SDK version.

arremebg, 2023-11-14:
Thanks for the quick response! We are using 3.8.2 and unfortunately we cannot migrate the entire codebase to 3.6 :( I

t's strange because we have been using this version for a year now and this never happened.