stereolabs/zed-sdk #475: All accelerometers have values of 0，When I tried to get IMU data from SVO

P970727, 2022-04-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I can't get complete IMU data from SVO. All accelerometers have values of 0. The data of SVO comes from ZED Explorer.exe with zed2 .

### Steps to Reproduce

1、Use the ZED Explorer.exe to record SVO
2、Modify the code in ZED SDK\samples\svo recording\export\python\svo_export.py，Adding code for reading IMU
3、The added code is as follows：
           sensors_data = sl.SensorsData()
       #The following is in the while loop------
           zed.get_sensors_data(sensors_data, sl.TIME_REFERENCE.IMAGE)
             # Extract IMU data
            imu_data = sensors_data.get_imu_data()
            angular_velocity = imu_data.get_angular_velocity()
            linear_acceleration = imu_data.get_linear_acceleration()

### Expected Result

The value of accelerometer is not 0

### Actual Result

0The value of accelerometer is 0

### ZED Camera model

ZED2

### Environment

```shell
os:win10
cpu:corei5
gpu：nvidia GeForce MX250
zed sdk:3.7.1
```

### Anything else?

no

obraun-sl, 2022-04-08:
Hi,

Just tested with 3.7.2 and works as expected. 

--> linear_acceleration returns correct values (norm = 9.8)
--> angular_velocity is at 0 as expected because the SVO contains only data that have a meaning in image framerate (i.e. accelerometer + magnetometer)

closed 2022-05-14 by github-actions[bot]

jujumo, 2022-09-27:
Same pb here, with version 3.7 (docker)

```
ack = self.cam.get_sensors_data(sensors_data, sl.TIME_REFERENCE.CURRENT)
```

always returns an error. I also tried with time reference image.

tutunarsl, 2024-09-15:
Hello @obraun-sl ,

Sorry for reviving a long-dead issue. I can see that above you mentioned that angular_velocity coming from the gyro is expected to be 0 when a SVO file is replayed. 

Is there a way to interfere with this behavior? Can you point me to the right direction to change this. This is currently diminishing the value of the sensor for me.

obraun-sl, 2024-09-27:
Hi, 
Sorry for the late reply. 

Since then, we have release SVO"Gen2" (which write *.svo2 files) where raw imu data are stored and available through the SDK 
(compared to SVOGen1 which only contains fused IMU data)

tutunarsl, 2024-09-27:
Thank you for your reply! Much appreciated. From what I see with[ ZED SDK version 4.1](https://community.stereolabs.com/t/zed-sdk-v4-1-is-out/4915) `.svo2` recording got into place. I am happy to upgrade. Before doing so, 

- I have a ZED2i instead of a ZEDX. Does this hinder my ability to use .svo2 ? 
- My integration is based on ROS1, is the `.svo2` format have special requirements?

Once again thank you for your support!