stereolabs/zed-sdk #479: New SDK can't build the sample.

459737087, 2022-04-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

in /zed/samples/tutorial/ttutorial-3-depth sensing/build.
when I cmake ..
It reported.
[image]

### Steps to Reproduce

mkdir build
cd build
cmake ..

### Expected Result

goes right

### Actual Result

bug

### ZED Camera model

ZED

### Environment

```shell
os:ubuntu 20.04
zed: 3.7
cuda: 11.5
```

Myzhar, 2022-04-24:
Did you update from an older SDK version? In this case you must clean the CMake cache by deleting the content of your build folder.
Then you must call the `cmake ..` command again and finally recompile.

459737087, 2022-04-24:
Hello @Myzhar ! the SDK version is the newest , and I deleted the build folder and recompile , but this problem still generate.

Myzhar, 2022-04-24:
Have you tried to install the BLAS libraries?
```
sudo apt-get update
sudo apt-get install -y libopenblas-dev
```

PS Always copy and past the full error log, screenshot are not a good way of providing error information,

459737087, 2022-04-24:
Hi! I have tried to install BLAS libraries, that problem has solved, but a new problem raised.
[CMakeError.log](https://github.com/stereolabs/zed-examples/files/8549336/CMakeError.log)
[CMakeOutput.log](https://github.com/stereolabs/zed-examples/files/8549337/CMakeOutput.log)

obraun-sl, 2022-04-25:
Hi,
You need to add pthread in the target link libraries list
https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/cpp/CMakeLists.txt#L39

It should be `TARGET_LINK_LIBRARIES(${PROJECT_NAME} ${ZED_LIBS} pthread)`

459737087, 2022-04-26:
Thank u! I have solved it.

closed 2022-04-26 by obraun-sl