stereolabs/zed-sdk #482: ZED yolov4 cpp example not compiling with Xavier NX and ZED 2 camera

abbaasalif, 2022-05-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi Team,
I am trying to compile the YOLOv4 object detection exxample and I get the following errors. I understand that we need to install opencv from source but its causing issues with my whole environment breaking. Please help me out and guide me through the right way.

### Steps to Reproduce

Steps done:
mkdir build
cd build
cmake ..
make 
Then it is causing this error while compiling.

### Expected Result

Should run fine

### Actual Result

Scanning dependencies of target opencv_dnn_zed
[ 33%] Building CXX object CMakeFiles/opencv_dnn_zed.dir/src/GLViewer.o
[ 66%] Building CXX object CMakeFiles/opencv_dnn_zed.dir/src/main.o
/home/morselabrx1/object_detection_v4/src/main.cpp: In function ‘int main(int, char**)’:
/home/morselabrx1/object_detection_v4/src/main.cpp:116:39: error: ‘DNN_BACKEND_CUDA’ is not a member of ‘cv::dnn’
  116 |     net.setPreferableBackend(cv::dnn::DNN_BACKEND_CUDA);
      |                                       ^~~~~~~~~~~~~~~~
/home/morselabrx1/object_detection_v4/src/main.cpp:117:38: error: ‘DNN_TARGET_CUDA’ is not a member of ‘cv::dnn’
  117 |     net.setPreferableTarget(cv::dnn::DNN_TARGET_CUDA);
      |                                      ^~~~~~~~~~~~~~~
CMakeFiles/opencv_dnn_zed.dir/build.make:86: recipe for target 'CMakeFiles/opencv_dnn_zed.dir/src/main.o' failed
make[2]: *** [CMakeFiles/opencv_dnn_zed.dir/src/main.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/opencv_dnn_zed.dir/all' failed
make[1]: *** [CMakeFiles/opencv_dnn_zed.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

### ZED Camera model

ZED2

### Environment

```shell
OS:  Nvidia Jetpack 4.6.1-b110
Device: Nvidia Jetson Xavier NX
ZED API version: ZED SDK 3.7.2
```

obraun-sl, 2022-05-09:
Hi,
You need to build opencv with CUDA and CUDA_dnn support to be able to use opencv_dnn with CUDA target (that will give you the best performances). 
You can use this repository to build opencv with the required support : 
https://github.com/mdegans/nano_build_opencv

abbaasalif, 2022-05-10:
Hey @obraun-sl, 

Thanks, I was able to successfully run it but just a catch that I wanted to register I was using GCC and g++ version 9 but OpenCV required version 8 and below. So just wanted to add that. Thanks for your timely response.

closed 2022-05-10 by abbaasalif