stereolabs/zed-sdk #484: how to get the camera parameters like other cameras(realsense)

anchao1997, 2022-05-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

when I use realsense cameras,I use "rs-enumerate-devices -c" to get all the camera parameters
so I wonder if there is the similar command or method to get the camera parameters
thanks

obraun-sl, 2022-05-13:
Hi,

you can use the sl::Camera::getDeviceList() static function to get the available cameras with their properties
See here : https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a1768d383f5e78746f58eda9d2c60b33e

anchao1997, 2022-05-16:
sorry
I explain that I want to get the exact rotation matrix,translation vector between camera1 to camera2, camera1 to imu. 
and the K matrix between every camera.
by the way I use the camera zed2
so can I use that website to get these parameters 
thank you very much

obraun-sl, 2022-05-16:
Ok. In this case, you can use this sample to extract camera parameters
https://github.com/stereolabs/zed-opencv-native

This will load the SNxxxx.conf file where those informations are stored.