stereolabs/zed-sdk #486: printing out the details of objects

urbansound8K, 2022-05-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

how to to print out the details of objects in terminal  in object_detection_birds_view.py?

like if I want to print something like this in terminal after I run object_detection_birds_view.py:
hight width, class, distance and confidence, please?

urbansound8K, 2022-05-25:
I got it but why 
print(object.label) whenever I print it out I get unknown only???

pQbas, 2023-02-28:
Use `print(object.raw_label)` instead.

closed 2025-09-26 by Myzhar