stereolabs/zed-sdk #492: Yolov5 Custom Detection Tutorial script outdated (detection.py)

AdrienGenilec, 2022-06-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Function torch_thread() from detector.py try to pass unvalid argument 'map_location' to Yolov5 function attempt_load()

### Steps to Reproduce

1.python detector.py
2.
3.
...

### Expected Result

no error

### Actual Result

line 100, in torch_thread
    model = attempt_load(weights, map_location=device)  # load FP32
TypeError: attempt_load() got an unexpected keyword argument 'map_location'

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 10
ZED SDK: 3.7.4
Yolov5 release: v6.1 (latest)
GPU: NVIDIA T500
CPU: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz   3.00 GHz
RAM: 32GB
```

### Anything else?

From my personnal little investigation it seems this script was up to date in older releases of yolov5 and that the 'map_location' argument should be replaced by 'device' which I tried and it seems to work although I am just a beginner... Also weights folder is missing in latest release of yolov5 so I get the following error if no --weights argument is given:
FileNotFoundError: [Errno 2] No such file or directory: '..\\weights\\best.pt'

I hope my fix is right and does not imply potentials mis-optimization of harware ressources (only 2% GPU usage seems low).

Best regards,
Adrien

urbansound8K, 2022-07-04:
did you solve it, please?

wasdee, 2022-07-05:
you could try to clone with the old commit of yolov5 like 6 month ago the same time the detect.py was edited

closed 2022-08-11 by github-actions[bot]

Marketos-Damigos, 2023-02-01:
@AdrienGenilec refer to #529 and our pull request #530 in order to use yolo v7 (which is almost a copy of v5) and get the example to work.