stereolabs/zed-sdk #501: ZedSdk 3.7.5 with Cuda 11.6.2 or 11.7.0 does not work properly

John-Gdi, 2022-07-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When we downlown the lastest 3.7.5 sdk with cuda 11.6.2 or 11.7，the ZED Depth Viewer.exe will crash, but the 3.6.5 works fine.

### Steps to Reproduce

1. Install CUDA 11.6.2 or Cuda 11.7
2. Install ZED SDK 3.7.5 for windows
3. Open the  `ZED Depth Viewer.exe` in tools directory.
4. Stop working.
...

### Expected Result

We can see the depth viewer result as 3.6.5 sdk version

### Actual Result

The app crashes when launchered

[image]

### ZED Camera model

ZED2i

### Environment

```shell
OS: win11
CPU:i7-11700 @ 2.5GHz
GPU:NVIDIA RTX 3060
ZED SDK Version： 3.7.5
CUDA version：11.6.2 or 11.7.0
```

johnsmakar, 2022-07-15:
We receive:
```
could not load library cudnn_cnn_infer64_8.dll. error code 193
``` 
When running the body_tracking sample, but our ZED Depth Viewer.exe works

John-Gdi, 2022-07-15:
when we run python demo, it down load cudnn related dlls, and everything looks right now.

closed 2022-07-15 by John-Gdi

johnsmakar, 2022-07-15:
Which python demo? Do you mean `get_python_api.py`?

John-Gdi, 2022-07-15:
> Which python demo? Do you mean `get_python_api.py`?

Before you run the samples with python in the `ZEDSDK_ROOT_DIR`, you need to run `get_python_api.py` for installing necessary dependencies.

When you first run body_tracking, it will download and install AI_model, please make sure your network can access stereolabs.com directly.

wuxiangchao, 2022-08-12:
@John-Gdi 
when i run
```
python body_tarcking.py
``` 
with the following exception:

```
Running Body Tracking sample ... Press 'q' to quit
freeglut (foo):  ERROR:  Internal error <FBConfig with necessary capabilities not found> in function fgOpenWindow
```