stereolabs/zed-sdk #502: version jetson 4.4 v3.7.5  cannot save area map

shlomi-amitai, 2022-07-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When moving to version 3.7.5. and traying to save map using zed.disable_positional_tracking(self.outMapFile) map is not saved.

### Steps to Reproduce

1. run positional tracking 
2.  tracking_params = sl.PositionalTrackingParameters()
 3.     tracking_params.area_file_path = inMap
            status = zed.enable_positional_tracking(tracking_params)
3.

No file saved
...

### Expected Result

saved map file

### Actual Result

Nothing happens.
in version 3.7.4. a map is saved

### ZED Camera model

ZED2i

### Environment

```shell
Jetson Linux jetpack 4.4
ZED SDK 3.7.5 for jetpack 4.4
```

Myzhar, 2022-07-17:
Hi @shlomi-amitai 
The code that you posted is correct to open an existing area map file.
To save the area map you can call the [saveAreaMap](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#afe82ca1a5a374233a0f28bf7ba747674) function or call the [disablePositionalTracking](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#ab30eefa02528b9c95d3a44aedb35ae2c) function

shlomi-amitai, 2022-07-17:
I'm sorry, pasted the wrong code, 
that' what I'm doing
zed.disable_positional_tracking(self.outMapFile)
and yet , it is not saved.
When switching to 3.7.4, it works (but then other issues come up...)

Myzhar, 2022-07-17:
We are going to test this odd behavior.
Can you try to use the saveAreaMap function to test if it works?

shlomi-amitai, 2022-07-17:
tried this as well, still not dumping the file.

closed 2022-08-23 by github-actions[bot]

YaroslavShchekaturov, 2023-03-13:
Faced the same issue