stereolabs/zed-sdk #503: initial_world_transform() and zed.get_position() have different results.

jinho0309, 2022-07-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

  c_translation = sl.Translation()
    c_translation.init_vector(0.039138934265215745, 0.33897992297324625, 3.7656952005014612)
    c_rotation = sl.Rotation()
    c_rotation.set_rotation_vector(0.05210374780239872, 2.6160165564830247, -1.37881240459529)
    c_transform = sl.Transform()
    c_transform.init_rotation_translation(c_rotation, c_translation)
 

    c_pose = sl.Pose()
    c_pose.init_transform(c_transform)
    rotation = c_pose.get_rotation_vector()
    print(rotation)
   

    positional_tracking_parameters = sl.PositionalTrackingParameters()
    positional_tracking_parameters.set_as_static = True
    positional_tracking_parameters.enable_area_memory= False
    positional_tracking_parameters.set_initial_world_transform(c_transform)
    zed.enable_positional_tracking(positional_tracking_parameters)

    t_transform = positional_tracking_parameters.initial_world_transform()
    print(t_transform.get_rotation_vector())

    if zed.grab() == sl.ERROR_CODE.SUCCESS : # Grab an image and computes the tracking
        camera_pose = sl.Pose()
        zed.get_position(camera_pose, sl.REFERENCE_FRAME.WORLD)
        rotation = camera_pose.get_rotation_vector()

        print(rotation)

### Steps to Reproduce

.

### Expected Result

,

### Actual Result

result :
[ 0.05210375  2.61601663 -1.37881243]
[ 0.05210375  2.61601663 -1.37881243]
[-0.34887791  2.80270791  0.80921131]

Why diffrent?

### ZED Camera model

ZED2

### Environment

```shell
OS : Winw
```

obraun-sl, 2022-07-18:
Hi,
This is expected as the initial world transform is adjusted with the IMU (gravity only) orientation of the camera to match the real world pose. 
There will be a setting in a future version to be able to de-activate that.

closed 2022-08-24 by github-actions[bot]

haavarpb, 2022-10-10:
Hi, sorry for necroing the issue but I just encountered the same phenomenon.

Following is copied from [PositionalTrackingParameters specification](https://www.stereolabs.com/docs/api/structsl_1_1PositionalTrackingParameters.html):

>bool set_as_static
>
>This setting allows you define the camera as static. If true, it will
 not move in the environment. This allows you to set its position using 
initial_world_transform. [Camera::getPosition()](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#aec3f07c489393fc818967f847a99acbb) will return the value set as initial_world_transform for the PATH, and identity as the POSE.
 
 Also, from [coordinate frames documentation](https://www.stereolabs.com/docs/positional-tracking/coordinate-frames/):
>World Frame
>
>Describing the position of the camera in absolute real-world space requires a reference point that remains stationary as the device moves. This global reference frame is called World Frame. If no initial parameters are set, the World Frame is located at the place where the ZED first started motion tracking, and is oriented in the direction where the device was looking. To get the position of the camera in real-world space, use getPosition(zed_pose, REFERENCE_FRAME::WORLD).

Firstly, I think the API should be cleaned up by renaming getPosition to getPose. As I understand it, in robotics, a pose is both a rotation and a position. The getPosition function is therefore misleading since the function actually gives you a pose. 

Secondly, I can't see any mentions of world frame being adjusted using the IMU. If this is behavior is indeed intended, the documentation should reflect that.