stereolabs/zed-sdk #507: Stereolabs.zed nuget package 3.7.11 is broken for spatial mapping mesh or point cloud.

gbriggs-trmb, 2022-08-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Using your spatial mapping example in C# from here:
https://github.com/stereolabs/zed-examples/tree/master/spatial%20mapping/basic/csharp 

When I run this using Stereolabs.zed Nuget package version 3.7.11, it is mostly broken:

1) You do not get the texture applied to the mesh. This line of code:
`zedCamera.ApplyTexture(ref mesh);`
prints this error into the console"
`[Mesh] [ERROR] Not enable, see sl::Camera::enableSpatialMapping for more informations.`

You do end up with a usable .obj mesh file, but without texture.  I double checked that `spatialMappingParameters.saveTexture = true;`

2) The point cloud file is invalid.  Run the sample in point cloud mode, it will create a ply file without errors, but if you try to open that ply file in CloudCompare, it complains about invalid data in the file.

I reverted back to Stereolabs.zed Nuget package version 3.7.0, and everything works fine (I get a mesh with texture, and I get a point cloud that opens in CloudCompare).

I believe the version 3.7.11 package is broken.

### Steps to Reproduce

1. Run the sample with Nuget 3.7.11 it is broken
2. Run the sample with Nuget 3.7.0 and it works.

### Expected Result

You should get an obj file with a texture, or you should get a ply that opens with CloudCompare

### Actual Result

You get an obj file with no texture, and you get a PLY file that is corrupted(?)

### ZED Camera model

ZED2i

### Environment

```shell
OS:  WIndows 10
Visual Studio 2022
```

Bvallon-sl, 2022-08-17:
Hi,

I don't have any issue opening ply files generated with the Csharp wrapper. I'm using MeshLab and not CloudCompare.

There is indeed a bug for the mesh generation, it will be fixed in the next release.
In the mean time, adding `spatialMappingParameters.useChunkOnly = true;` should fix it.

Best,
Benjamin Vallon

gbriggs-trmb, 2022-08-24:
Benjamin
Thank you for your reply. 

I will use your work around for the mesh generation and try the latest version, and watch for an update in the future.

I discovered the problem with your .ply files in CloudCompare.  They are missing a newline character at the end of the file, and apparently cloud compare does not like this. I 'fixed' this in my program by simply opening the .ply file after it is created and appending a newline at the end of it. You may consider adding this fix to your code as well.

Thank you
GB

closed 2022-09-30 by github-actions[bot]