stereolabs/zed-sdk #508: Install SDK with a given path

sunwell1994, 2022-08-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

The SDK should provide the option to install it in a given path instead of using the root permission to install it under /usr/local

### Use-Case

In the server without the root permission

Lj1212am, 2022-11-02:
Also having this issue, trying to get it working in docker

vini-at-potential, 2024-07-30:
Same, I my case it also happens that each user might need a different version of the SDK, so being able to choose where to install and have multiple versions installed would be very convenient.