stereolabs/zed-sdk #510: ZED Mapping not working as expected

Seye4, 2022-09-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I tried using the zed 2 camera to do 3d mapping using the sdk, Unity and also the ZEDfu, I couldn't get the same kind of result as shown in the website or on YouTube. The result I have has a lot of empty blank areas. I have tried to scan the environment slowly and also manage reflective surface and lighting of the environment but I still couldn't get a good result. I have also played with the settings available. Kindly find attached result of some of my scanning.

Can someone advise how to fix this issue?

### Steps to Reproduce

1. Used ZEDfu, scan has a lot of empty blank areas.
2. Used Unity spatial scanning setting has  a lot of empty blank areas.
3. Played with settings available in unity sdk and zedfu, same result
...

### Expected Result

Trying to achieve same result as shown in the website

### Actual Result

[image]

### ZED Camera model

ZED2

### Environment

```shell
WINDOWS 11
12th Gen Intel(R) Core(TM) i7-12700H   2.70 GHz
RAM 16.0 GB (15.7 GB usable)
NVIDIA GeForce RTX 3050 laptop GPU
```

Seye4, 2022-09-27:
Alternatively is it possible to just get a clay model result with just one monotonous color if that is easily achievable

hantupayung, 2022-09-27:
you can create a customized shader in opengl to give an accurate and complete mapping

closed 2022-11-03 by github-actions[bot]

IArchi, 2022-12-06:
Same behaviour here.