stereolabs/zed-sdk #511: SDK 3.7.7 Python error - cuDNN library not found

sumiboss95, 2022-10-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello

I installed the newest version of the software - ZED SDK 3.7.7 for Windows 10 and CUDA v11.7. I also installed Python as was described in Python development section of your website. When I try the tutorial examples with Visual Studio and C++ everything works as expected. However, when I try the Object detection and Body tracking examples in Python I get the following error:

[ZED][INFO] Logging level INFO
[ZED][INFO] [Init]  Depth mode: PERFORMANCE
[ZED][INFO] [Init]  Camera successfully opened.
[ZED][INFO] [Init]  Sensors FW version: 776
[ZED][INFO] [Init]  Camera FW version: 1523
[ZED][INFO] [Init]  Video mode: HD720@60
Object Detection: Loading Module...
CORRUPTED SDK INSTALLATION
[ZED][ERROR] [Object Detection]  cuDNN library not found
[ZED][ERROR] [ZED] [Object Detection]  CUDNN NOT FOUND

I tried uninstalling everything related to NVIDIA, and the ZED SDK itself, and then installing first the ZED SDK and followed the CUDA installation as performed by the SDK installer, but I get the same error. I tried uninstalling and reinstalling the SDK itself but I get the same problem. I tried downloading the cuDNN independently but still I get the same error again. I'm now not sure if this is a bug in the latest SDK version or if the problem is in something I did. I did not change anything in the Python code itself, so it is used exactly as provided by your examples on github. 

### Steps to Reproduce

1. Download the newest ZED SDK
2. It installs CUDA as well
3. Try running object_detection.py script from github tutorials

### Expected Result

I expect the program to work correctly and detect humans.

### Actual Result

I get the error I described:

Object Detection: Loading Module...
CORRUPTED SDK INSTALLATION
[ZED][ERROR] [Object Detection]  cuDNN library not found
[ZED][ERROR] [ZED] [Object Detection]  CUDNN NOT FOUND

### ZED Camera model

ZED2

### Environment

```shell
OS: Microsoft Windows 10 Pro Education   Version: 10.0.19042 Build 19042
CPU: Intel(R) Core(TM) i5-10300H CPU @ 2.50GHz   2.50 GHz
GPU: NVIDIA GeForce GTX 1650 Ti
Python version: Python 3.10.7
```

sumiboss95, 2022-10-05:
It apperas the Python version was 3.10.6 after all and after updating to 3.10.7 and reinstalling the Python api the program now works.

closed 2022-10-05 by sumiboss95

Myzhar, 2022-10-05:
Hi @sumiboss95 
please close the issue if you find a working solution to it.

sumiboss95, 2022-10-05:
It is now closed.