stereolabs/zed-sdk #512: Tutorial 2 - image capture doesn't work with C API

suncf2, 2022-10-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, I have to use ZED with C api.
After I build sl_zed_c.dll, I meet a error when I try to run the image capture sample code.
The error is when sl_retrieve_image succeeds, image_ptr doesn't contain a valid video frame which is expected a RGBA frame.

But ZED C++ SDK works perfect on the same computer.

### Steps to Reproduce

1. Build sl_zed_c.dll and sl_zed_c.lib with cmake-gui
2. Compile image capture sample
3. Run sample and every sl_xxx interface returns SUCCEED
4. After sl_retrieve_image returns, image_ptr doesn't contain right data (I expected RGBA)

### Expected Result

sl_retrieve_image returns a pointer which I can get RGBA data.

### Actual Result

sl_retrieve_image returns a pointer which is not as large as width * height * 4, and contains error data

### ZED Camera model

ZED Mini

### Environment

```shell
Win 10
11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
NVIDIA GeForce RTX 3060 Laptop GPU
ZED SDK 3.7.7
Cuda 11.7
```

Bvallon-sl, 2022-10-17:
Hi,

The retrieve_image function is retrieving a pointer toward a "sl::Mat" object, not the array of data.

If you want to have access to a specific value, you need to add the following lines of code : 

> 			int half_width = width / 2;
> 			int half_height = height / 2;
> 			//Get the pixel value of the center of the image
> 			sl_mat_get_value_uchar4(image_ptr, half_width, half_height, &value, SL_MEM_CPU);
> 
> 			printf("Pixel value of the center of the image [%i ,%i ,%i] \n", value.x, value.y, value.z);

Their are multiple support functions for sl::Mat manipulation in C, all starting with "sl_mat_"

For example, to get the height of the matrix : "sl_mat_get_height" .

Let me know if you still have issues.

Best regards,

suncf2, 2022-10-17:
> Hi,
>
> The retrieve_image function is retrieving a pointer toward a "sl::Mat" object, not the array of data.
>
> If you want to have access to a specific value, you need to add the following lines of [...]

Thank you for your quick reply!
Yes, everything works fine when I use "sl_mat"  functions.
Thanks very much!!

closed 2022-10-17 by suncf2

suncf2, 2022-10-17:
But anyway, please pay attention to "INTERFACE_API int mat_get_value_uchar4(int* ptr, int col, int raw, struct SL_Uchar4* value, enum SL_MEM mem);" function.
In this file "https://github.com/stereolabs/zed-c-api/blob/main/include/sl/c_api/zed_interface.h"

This function is **mat_get_value_uchar4**, not **sl_mat_get_value_uchar4**.
Is this a typing mistake?

Bvallon-sl, 2022-10-17:
Indeed, that's a mistake. Thanks for the report.

It will be fixed in the 3.8 SDK release (available very soon).

Best,