stereolabs/zed-sdk #514: ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type ---> error

SeongJun-Hwang, 2022-11-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello Developers.
The following issues have been identified: Please check

`C:\Users\guest_1\Documents\zed-examples\depth sensing\python>C:/Users/guest_1/.conda/envs/zed_test/python.exe "c:/Users/guest_1/Documents/zed-examples/depth sensing/python/depth_sensing.py"
Running Depth Sensing sample ... Press 'Esc' to quit
Traceback (most recent call last):
  File "c:/Users/guest_1/Documents/zed-examples/depth sensing/python/depth_sensing.py", line 51, in <module>
    viewer.init(len(sys.argv), sys.argv, camera_model, res)
    return __glutCreateWindowWithExit(str(title), _exitfunc)
ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type
`

### Steps to Reproduce

There's the same thing in all of Python's examples.

1. enter python example folder and launch main file 

### Expected Result

unable to run ----> ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type

### Actual Result

ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type

### ZED Camera model

ZED2i

### Environment

```shell
OS: window11
zed-sdk: 03-11-2022 lastest version
cuda: v11.7
python: 3.8
```

### Anything else?

Enter zed-examples\depth sensing\python\ogl_viewer\viewer.py

and modify next code

[line 235]
 glutCreateWindow("ZED Depth Sensing")  ------>  glutCreateWindow(b"ZED Depth Sensing")

then it will be work
If you have questions then please contact me
mail: seongjun.hwang@sisinc.co.kr

braoutch, 2022-11-03:
Hello, thank you for reporting this. Most users do not have this issue, as far as I know. Can you tell us what your pyOpenGL version is ?

alassagne-sl, 2022-11-03:
Hello, thank you for reporting this. Most users do not have this issue, as far as I know. Can you tell us what your pyOpenGL version is ?

closed 2022-12-09 by github-actions[bot]