stereolabs/zed-sdk #520: Depth Colorization

AlexanderKhazatsky, 2022-11-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I would like a built in procedure for turning depth readings into a color map. This is a useful feature on its own, but it also makes compression much easier. Ideally, this transformation would also be invertible as to recover the depth readings for a point cloud creation :)

P-yver, 2023-10-26:
Hi,
actually you can retrieve the depth as Usigned short, MEASURE::DEPTH_U16_MM. Then you can save it as png file (the depth is stored in millimeter as unsigned short).
Do you have example of your proposed format ?

katsunori-waragai, 2024-12-26:
I released a library to view disparity image.
I hope this is useful for your purpose.

https://github.com/katsunori-waragai/disparity-view

closed 2025-09-26 by Myzhar