stereolabs/zed-sdk #527: Is there the way to calculate distance/position to/of an object instead of pixel?

levmarki, 2022-12-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

The "common" way to detect the distance to small objects is to run the object detection on RGB, apply the detection to the depth and then filter it in some way (averaging/median/whatever). 
Is there the way to calculate distance to the middle/top/bottom of the object **only**? Without using the Depth Map?

### Use-Case

First of all, it will give us better results on small objects, it's much simpler to find registration between two objects (the same object from different cameras) instead of pixel calculation.
In addition, maybe it can be more calculage-efficient to calculate the distances to the objects of interest instead of whole frame?

P-yver, 2023-10-26:
Hi,
if you don't want to use the depth data you can compute the 3D position of the object if you have its 2D position within the left and right image, then use the stereo model to compute its 3D position.

levmarki, 2023-10-26:
Thank you for your reply.
The main question is if is there some way to match objects between the
cameras?

closed 2025-09-26 by Myzhar