stereolabs/zed-sdk #534: Memory Leak In SVO Reading

AlexanderKhazatsky, 2023-02-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, I am experiencing a memory leak which appears to have been reported a few years ago:https://github.com/stereolabs/zed-python-api/issues/111

Has there been any progress on this? I am using the most up to date SDK.

### Steps to Reproduce

Everything is described in that other doc

### Expected Result

Once you close the camera it should not take any more memory.

### Actual Result

Memory usage slowly grows over time as I load batches with the SVO functionality.

### ZED Camera model

ZED

### Environment

```shell
Linux
```

AlexanderKhazatsky, 2023-02-16:
Bump. Would really appreciate some info on what to do about this.

adujardin, 2023-02-16:
Hi,
The issue mentioned was fixed around ZED SDK 3.0. 

Could you give more details about the version you're using? Does the snippet in the old issue https://github.com/stereolabs/zed-python-api/issues/111 produce this result? Is it for ZED camera (or ZED M, ZED2 ?) SVO specifically? Which compression and image resolution was used to record the SVO? It may not be specific to the SVO but just in case, it may help pinpoint issues with specific configurations.

AlexanderKhazatsky, 2023-02-16:
Hi, I've been investigating this more today, I realized that the memory leak only happens in a specific instance.

I believe it occurs when we read from multiple SVO files concurrently. In other words, open up multiple SVO files, and read them all step by step (as to align frames from different angles).

I am currently working on this and will let you know if I find out more.

AlexanderKhazatsky, 2023-02-16:
I have an update, I was wrong about the cause. The memory leak is not present when reading SVO files written by a Zed MINI, but is present when reading SVO files written by a Zed 2. Even when reading just one file at a time.

AlexanderKhazatsky, 2023-02-17:
Okay, looks like the memory look happens with Zed 2's and not Zed Mini's even when they are reading at the same resolution / frame rate

AlexanderKhazatsky, 2023-02-17:
If I were to start data collection on this SDK version, will I be able to read the SVO files on a later SDK that fixes this issue?

Myzhar, 2023-02-17:
@AlexanderKhazatsky Yes, an SVO recorder with older SDK versions will be always compatible with newer SDK versions, while the contrary cannot be always valid.
@adujardin asked for the SDK version that you are using, but you have not yet provided this important information.

AlexanderKhazatsky, 2023-02-17:
I am on the most recent SDK: 3.8.2 for CUDA 11.1 - CUDA 11.7

adujardin, 2023-02-17:
@AlexanderKhazatsky Thanks for the additional information, we'll investigate on our side to try to replicate this. In the meantime, if you find other elements don't hesitate to share them. 
Would it be possible to share one of your SVO file you have that triggers this issue? You can send it privately to support@stereolabs.com if needed.

AlexanderKhazatsky, 2023-02-17:
Here's the data, along with an example graph of the memory leak:
[recordings.zip](https://github.com/stereolabs/zed-examples/files/10772033/recordings.zip)
[image]

AlexanderKhazatsky, 2023-02-17:
The graph may be a bit different because it's from me loading all 3 recordings at once across different videos.

lilianmallardeau, 2023-02-21:
Hi @AlexanderKhazatsky,
I'm currently investigating on the issue. Are you using the SDK with the Python API or in C++?

AlexanderKhazatsky, 2023-02-21:
Just to clarify, I'm using the Python API

lilianmallardeau, 2023-02-23:
Hi @AlexanderKhazatsky, which distribution (and version) are you using, and what is the version of your Linux kernel?

AlexanderKhazatsky, 2023-02-23:
Here's the output of uname -a:
- Linux iris-mobile 5.15.0-60-generic #66~20.04.1-Ubuntu SMP Wed Jan 25 09:41:30
TC 2023 X86_64 X86_64 X86_64 GNU/Linux

Here's the output of pip freeze | grep pyzed (Which I think is what you were asking for? Not totally sure):
- pyzed @ file: ///tmp/selfgz18515/pyzed-3.8-c37-c37m- linux_X86_64.Whl

If you're asking about the SDK version, I'm using the most recent version.

lilianmallardeau, 2023-02-24:
Thanks. We're currently investigating on the issue. If you need a fix ASAP however, you can switch to Ubuntu 22.04 since the problem doesn't seem to happen on it.

AlexanderKhazatsky, 2023-02-25:
This solves the issue on our side! However I do notice that reading files seems slower? Previously I was able to read through 100 files in ~12 seconds, now it takes about a minute. Is this expected?

lilianmallardeau, 2023-03-01:
Hi @AlexanderKhazatsky,

We found the cause of the issue. It isn't actually a memory leak, but this is due to how glibc manages memory under the hood.

When `malloc` (or friends, `calloc`, `realloc` etc... and the C++ `new` operator) is used, it usually allocates memory from the process' heap and returns a pointer to this area of memory. If the heap is not big enough or if there's no free contiguous memory chunk of the requested size available, `malloc` makes a system call to `brk`/`sbrk` to ask the kernel to increase the size of the heap, and then allocates the requested memory and returns a pointer to it.

When using `free` (or `delete`) however, the memory area is marked as available again in the heap for the current process, but this generally does *not* return the memory back to the OS. The chunk of memory is kept for the process for future allocations. Some memory is released if the amount of free memory *at the end* of the heap exceeds a predefined threshold `M_TRIM_THRESHOLD` (which is 128kB by default). Free chunks of memory "sandwiched" between two other allocated chunks cannot be released back to the OS. When this happens, this is called heap fragmentation.

Heap fragmentation is when the heap contains multiple free memory chunks (which cannot be returned to the OS) between some other allocated chunks:
X...XXX.....X...X
(the allocated memory chunks are represented as X and the free ones as ...)
Those free memory chunks can (and will, if they're big enough) be used for future allocations in the process, but cannot be released to the OS, until all the following memory chunks in the heap are freed. For example, here, if the last X is freed, and ...X (the 3 dots plus the X) is more than 128kB, this memory chunk will be released to the OS because `free` will automatically call `brk`/`sbrk` to reduce the heap size appropriately.

Now, actually, in glibc, when `malloc` is requested an amount of memory bigger than `M_MMAP_THRESHOLD` (128kB by default), and if a contiguous memory chuck of the requested size is not available in the heap, it uses the `mmap` syscall to get a memory mapping of the requested size from the OS, instead of using `brk`/`sbrk` to increase the size of the heap. In this case, the memory in released back to the OS directly when it is freed.

So, to come back to the original issue, I see 2 possibilities to force the process to give back unused memory to the system:
- `malloc` actually allows to tweak its behaviour via environement variables. See `mallopt(3)` for all the details, but in a nutshell:
	- Setting the environment variable `MALLOC_TRIM_THRESHOLD_` allows to change the `M_TRIM_THRESHOLD` value.
	- Setting the environment variable `MALLOC_MMAP_THRESHOLD_` allows to change the `M_MMAP_THRESHOLD.` value.

	Now you have to understand that setting those values is a tradeoff between memory usage and speed: using a low value for `M_TRIM_THRESHOLD` will reduce memory usage because it will give back memory to the OS more frequently, but it will increase the number of system calls. Setting `M_TRIM_THRESHOLD` to 0 will force the process to always give back the unused memory at the end of the heap to the OS.

	The manual says:

        Modifying M_TRIM_THRESHOLD is a trade-off between increasing the number of system calls (when the parameter is set low) and wasting unused memory at the top of the heap (when the parameter is set high).

	You can try different values and see what works best for you. But in your case, you want to try values lower than 128*1024, which is the default.

- Use an alternative implementation of `malloc` and friends, with improved memory management, for example [`tcmalloc`](https://github.com/google/tcmalloc). I tried it on the SDK and it seems to work pretty well. It uses less memory and is faster. It's not planned to integrate it into the SDK but maybe we will discuss it if it shows to have real benefits. Don't hesitate to give some feedback about it if you use it with the SDK.
To use it with your current program, install it first (`sudo apt install libtcmalloc-minimal4`), and use the `LD_PRELOAD` environment variable to run your program with it:

        LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libtcmalloc_minimal.so.4 ./your_program

	It will override the implementations of `malloc`, `calloc`, `new` etc.
	See https://goog-perftools.sourceforge.net/doc/tcmalloc.html for more infos and docs.

The fact that the behaviour is different between Ubuntu 22.04 and Ubuntu 20.04 is probably due to an improvement in the memory management in glibc (libc6 2.31 on ubuntu 20.04 vs libc6 2.35 on ubuntu 22.04).

If you want to learn more about this, here are some ressources I found helpful to understand everything:
- http://www.cplusplus-soup.com/2010/01/freedelete-not-returning-memory-to-os.html
- https://www.reddit.com/r/coding/comments/2ditbr/comment/cjpwfff/?utm_source=share&utm_medium=web2x&context=3
- https://www.gnu.org/software/libc/manual/html_node/Freeing-after-Malloc.html#Freeing-after-Malloc
- https://stackoverflow.com/questions/2215259/will-malloc-implementations-return-free-ed-memory-back-to-the-system
- https://stackoverflow.com/questions/48358229/how-can-i-get-a-guarantee-that-when-a-memory-is-freed-the-os-will-reclaim-that

AlexanderKhazatsky, 2023-03-16:
Hello, unfortunately, it seems I’m actually still experiencing a roughly linear increase in memory usage on ubuntu 22. Could I send you guys my files and see if you are able to reproduce this on your end?

AlexanderKhazatsky, 2023-03-16:
It’s also now happening with every type of camera.

lilianmallardeau, 2023-03-27:
Mhhh, strange. Have you tried using tcmalloc?

closed 2023-05-03 by github-actions[bot]