stereolabs/zed-sdk #539: Zed's Crash When Closing and Reopening Cameras

AlexanderKhazatsky, 2023-03-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I have a list of cameras, if I continue opening and closing them, it eventually fails and my terminal outputs something similar to "can't claim interface 0: -6"

My purpose for doing this is that I want to collect data in a low resolution, but extrinsically calibrate a single camera at a time with high resolution. I only have 3 cameras on at once, and this issue happens frequently.

### Steps to Reproduce

Make a while True loop where you turn cameras on and then off.

### Expected Result

Procedure should be safe to repeat

### Actual Result

Eventually camera's cannot be turned back on again.

### ZED Camera model

ZED2

### Environment

```shell
I am using the most recent SDK, ubuntu 22, and have zed minis and zed 2's.
```

AlexanderKhazatsky, 2023-03-16:
I wanted to bring attention to this issue again whenever you guys have the chance. The memory leak issue I mentioned in another thread is more blocking than this though :)

Myzhar, 2023-04-04:
Noted. We are working on this

AlexanderKhazatsky, 2023-04-06:
Just to check- are you guys able to reproduce the issue?

AlexanderKhazatsky, 2023-05-07:
Hey! Can you guys please address this.

closed 2023-06-13 by github-actions[bot]