stereolabs/zed-sdk #544: Playback inconsistently ending one frame before

rodrigoberriel, 2023-03-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The playback, via python, sometimes ends one frame before the expected. The behavior is not deterministic, but I was able to consistently reproduce it in many PC configurations with several videos recorded from different ZED versions (ZED 1, ZED 2, and ZED 2i).

### Steps to Reproduce

To reproduce it, I used the playback snippet provided in the [official website](https://www.stereolabs.com/docs/video/using-video/#video-playback), with small changes to print additional info:

```python
# FROM: https://www.stereolabs.com/docs/video/using-video/#video-playback
import sys
import pyzed.sl as sl

# Set SVO path for playback
input_path = sys.argv[1]
init_parameters = sl.InitParameters()
init_parameters.set_from_svo_file(input_path)

# Open the ZED
zed = sl.Camera()
err = zed.open(init_parameters)

svo_image = sl.Mat()
while True:
    if zed.grab() == sl.ERROR_CODE.SUCCESS:
        # Read side by side frames stored in the SVO
        zed.retrieve_image(svo_image, sl.VIEW.SIDE_BY_SIDE)
        # Get frame count
        svo_position = zed.get_svo_position()
    elif zed.grab() == sl.ERROR_CODE.END_OF_SVOFILE_REACHED:
        print("SVO end has been reached.")
        print(f"Current SVO position: {svo_position + 1}")
        print(f"Nb of frames: {zed.get_svo_number_of_frames()}")
        print(f"Nb of frames dropped: {zed.get_frame_dropped_count()}")
        exit()
```

### Expected Result

I would expect the "Current SVO position" to be equal to the "Nb of frames". The number of frames dropped is reported just in case as well.

### Actual Result

But if I run it 10 times, they are different in a few of them:

```bash
for i in {1..10}; do python3 playback.py HD1080_SN23944511_16-20-25.svo; done

SVO end has been reached.
Current SVO position: 182
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 182
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 182
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 183
Nb of frames: 183
Nb of frames dropped: 0
SVO end has been reached.
Current SVO position: 182
Nb of frames: 183
Nb of frames dropped: 0
```

### ZED Camera model

ZED2

### Environment

```shell
I tried on several environments, locally and in the cloud.

OS: Ubuntu 20.04 and 22.04
GPU: GTX 1050 Ti, Tesla T4, and Tesla V100.
ZED SDK VERSION: v3.7.6 and v3.8.0
Other info: I tried it in docker containers and directly on the host.
```

### Anything else?

I tried to disable depth at runtime (passing `depth_mode=sl.DEPTH_MODE.NONE` to `sl.InitParameters(...)`), which I was expecting to improve the situation, but exactly the opposite happens: the inconsistency becomes far more frequent.

I also tried videos from different ZED versions: ZED, ZED 2, ZED 2i. All of them presented the same issue.

Is this expected?

AlexanderKhazatsky, 2023-04-03:
I have ran into this issue as well

rodrigoberriel, 2023-04-14:
Update: I sent this issue to support@stereolabs.com and the response was:

> We are aware of this bug and will fix it.
> If it's a big issue for you, the best would be to use 3.8.2.

In this context, I reproduced the steps provided in the description of this issue using SDK 3.8.1 and 3.8.2 and **the problem persists**. It happens as frequently as I reported before.

As an additional feedback, I also tried using SDK 4.0.1 (early access) and the issue there is different: the playback _consistently ends one frame earlier_.

I asked them whether there is anything else to try or if we should wait for a fix. I'll post an update in case they answer to directly to me.

closed 2023-05-21 by github-actions[bot]