stereolabs/zed-sdk #547: What is the downsizing procedure?

AlexanderKhazatsky, 2023-04-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I would like to convert SVO files to MP4 files, and at a later time down size them in such a way that is identical to passing in the resolution parameter. What is the exact approach used for downsizing images so that I can replicate it without any distribution shift to my model?

Furthermore, it would be awesome if there was a built in way to convert SVO files to MP4 files without having to compress (live to svo), decompress, and then recompress into a new MP4 file.

### Use-Case

Converting from SVO files to a more easily readable and supported format.

obraun-sl, 2023-04-13:
Hi,

You can use the sample for svo export here : 
https://github.com/stereolabs/zed-sdk/tree/master/recording/export/svo

You just need to create a single mode left only. 
You can resize the image using cv::resize() and adjust the resolution when creating the opencv videowriter
 Here : https://github.com/stereolabs/zed-sdk/blob/master/recording/export/svo/cpp/src/main.cpp#L136

`Furthermore, it would be awesome if there was a built in way to convert SVO files to MP4 files without having to compress (live to svo), decompress, and then recompress into a new MP4 file.`

That is not possible. SVO file contains RAW YUYV side by side image and the MP4 file will contain RGB Left image. So it cannot be the same encoding. it will always require re-encoding.

AlexanderKhazatsky, 2023-04-13:
Hi! The cv2.resize function aside, I wanted to know what resizing scheme was used when we pass in the Zed resolution to the reader functions. Basically, I want to know what resizing parameters to use with cv2.resize so that the resulting images are the same.

obraun-sl, 2024-09-27:
We use bilinear sampling inside the SDK for direct resizing