stereolabs/zed-sdk #551: Segmentation fault error when optimizing AI model with ZED SDK installer

ggjj321, 2023-04-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When attempting to optimize an AI model using the 

```ZED_SDK_Ubuntu20_cuda11.8_v4.0.1.zstd.run```

 installer for the ZED SDK, a segmentation fault error occurs. However, this issue is not present when using another installer like
```ZED_SDK_Ubuntu20_cuda12.1_v4.0.1.zstd.run```.

### Steps to Reproduce

```
./ZED_SDK_Ubuntu20_cuda11.8_v4.0.1.zstd.run
```

### Expected Result

[image]

### Actual Result

[image]

### ZED Camera model

ZED2

### Environment

```shell
OS : Ubuntu20.04
CPU : Intel
GPU : NVIDIA RTX A4000
installer version : ZED_SDK_Ubuntu20_cuda11.8_v4.0.1.zstd.run
```

obraun-sl, 2023-04-13:
The installer launch ZED_Diagnostic -aio at the end for all the models but it's not mandatory.
The fact to launch all the optimization might not be supported, depending on the memory available. 

You can skip this step and launch manually the optimization of AI model. 
Use ZED_Diagnostic with the specific parameter to optimize the model you want.
Ex : ZED_Diagnostic -ais <X> , where X is the sl::DETECTION_MODEL you want to optimize

ggjj321, 2023-04-24:
OK, thank you.

closed 2023-04-24 by ggjj321