stereolabs/zed-sdk #558: CUDA 12.1 not being detected

eezstreet, 2023-04-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using ZED SDK 4.0.2 (current latest) and it is not detecting CUDA in the diagnostic.

I tried reinstalling the ZED SDK and that didn't work. It requires CUDA 12.1 - I installed it, and it didn't work. I uninstalled both CUDA 11.7 (which was installed previously) and CUDA 12.1 and then reinstalled CUDA 12.1. It wasn't detected. I reinstalled the ZED SDK and it is still not detected.

What do I need to make it show up?

### Steps to Reproduce

1. Install ZED SDK 4.0.2 over existing ZED SDK 4.0.1 installation (with CUDA 11.7 installed)
2. CUDA won't be detected.

### Expected Result

It should detect the CUDA 12.1 toolkit.

### Actual Result

It doesn't detect the toolkit.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 10
CPU: i7-4790
GPU: NVIDIA GeForce GTX 1060 6GB
ZED SDK Version: 4.0.2
```

eezstreet, 2023-04-28:
Just adding here that a colleague of mine is having the same issue. He's using a much more recent GPU than I am.

I uninstalled the newer version and attempted to downgrade to 4.0.1 but it didn't work - the GPU is no longer detected and CUDA is still not detected. If we could get some word ASAP on how to fix this issue that would be good.

ahumphries2, 2023-05-02:
Same issue. nvcc --version confirms that CUDA 12.1 installed, but Zed Diagnostic tool tells me that I don't have 12.1 CUDA.

asjalrashidi01, 2023-05-03:
I am facing the same issue. I am on a fresh install of ZED SDK v4.0.2, which does not recognize CUDA v12.1.0. The camera works fine in the Explorer and all other Diagnostic tests pass. I am trying to utilize the Unity plugin which also gives the error SDK not installed I presume due to the CUDA error. nvcc --version also confirms the CUDA install for me as mentioned above as well.

ghkd3930, 2023-05-04:
Same issue. The explore or unity packages are work. I forcibly tried to optimize AI through cmd, but an error saying that the path could not be found came out. I tried using cuda 11.8, but I get an error saying that it can't find the path too.

[image]

P-yver, 2023-05-05:
Hi,
the cuda detection issue will be fixed in the upcoming 4.0.3 release.
This is simply an issue of the ZED Diag tool, not directly from the SDK.
Even if the Diag Tool report an issue you should be able to run tutorials or samples without issues.

ghkd3930, 2023-05-08:
> Hi, the cuda detection issue will be fixed in the upcoming 4.0.3 release. This is simply an issue of the ZED Diag tool, not directly from the SDK. Even if the Diag Tool report an issue you should be able to run tutorials or samples without issues.

hello. @P-yver.  I want to install AI in NEURAL mode. However, it is impossible to install using DIAGNOSTIC as above, and it is also impossible to directly key in Depth Mode. Isn't this the same problem? Installing Body Skeleton AI is not a problem.

closed 2023-06-13 by github-actions[bot]