stereolabs/zed-sdk #559: Can't get the resolution value

pfcouto, 2023-04-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am in Windows and using the SDK 3.8 and python. I am trying to run an example code `recording\export\svo\python\svo_export.py` and it outputs an error saying that it can't get the resolution. I tried to look for documention on the SDK and didn't find any, so if you know where it, or if it exists please provide it in the commnets. 

The error is in the print bellow, thank you for your help!

[image]

### Steps to Reproduce

1. ZED SDK 3.8
2. Try to run the python file [here](https://github.com/stereolabs/zed-sdk/blob/master/recording/export/svo/python/svo_export.py)
3. I am using the command `python .\svo_export.py "C:\Users\pedro\Documents\ZED\HD1080_SN30952243_12-33-49.svo" "C:\Users\pedro\Documents\ZED\Exports\" 4`

### Expected Result

Expected the code to run

### Actual Result

The code exited with the error `AttributeError: 'pyzed.sl.CameraConfiguration' object has no attribute 'resolution'`

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 11
CPU: x86
SDK version: 3.8
```

### Anything else?

Hope you can help me out

drewjgray, 2023-05-03:
I got the same issue.

P-yver, 2023-05-05:
Hi,
this is due to the fact that the repo is updated for the ZED SDK4.0.
In your case you are using the 3.8 API, check into the 3.8 TAG:
https://github.com/stereolabs/zed-sdk/blob/v3.8/svo%20recording/export/python/svo_export.py

closed 2023-06-11 by github-actions[bot]