stereolabs/zed-sdk #566: Lat Lon to UTM conversion bug

schaelo, 2023-05-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The latlng2utm function of sl::GeoConverter is not returning the correct UTM information. Not in degree nor in rad as inputs.

### Steps to Reproduce

1. Input positional information in function
2. get incorrect results

### Expected Result

Correct UTM coordinates

### Actual Result

UTM coordinates close to equator

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
ZED SDK: 4.0.2
```

schaelo, 2023-05-26:
`sl::LatLng TestPoseLatLon; `
`sl::UTM TestPoseUTM; `
`TestPoseLatLon.setCoordinates(double(0.86371764937), double(0.1477689993), double(90));`
 `sl::GeoConverter::latlng2utm(mLastLatLongPose, TestPoseUTM);`
The lon and lat values are in radians 
Expected are UTM coordinates Easting: 461363.603  Northing: 5481774.103 32U

TanguyHardelin, 2023-05-26:
Hello @schaelo 
Thanks for your feedback ! Indeed it's a bug from the SDK. It will be fixed in the next release !

closed 2023-07-01 by github-actions[bot]