stereolabs/zed-sdk #574: Undefined symbol when importing pyzed

TheoCarme, 2023-06-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I try to run this [code](https://github.com/TheoCarme/PixelwiseRegression21Joints/blob/master/rgb_hands_tracking.py) in a pipenv virtual environment, I get this error :
```
    import pyzed.sl as sl
ImportError: /home/labcom/.local/share/virtualenvs/PixelwiseRegression21Joints-p2Yj0yPh/lib/python3.10/site-packages/pyzed/sl.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZN2sl24SpatialMappingParametersC1ENS0_18MAPPING_RESOLUTIONENS0_13MAPPING_RANGEEibbbNS0_16SPATIAL_MAP_TYPEEi
```
And I now it can work since I deleted my precedent environment (like a fool) with which it was working fine.

### Steps to Reproduce

1. pipenv shell
2. pipenv install numpy
3. pipenv install opencv-python
4. pipenv install mediapipe
5. pipenv install pyzed-4.0-cp310-cp310-linux_x86_64.whl
6. pipenv run python rgb_hands_tracking.py

### Expected Result

I was supposed to get the stream of the camera with the tracking of the hands on it.

### Actual Result

I get the error mentioned above : 
```
    import pyzed.sl as sl
ImportError: /home/labcom/.local/share/virtualenvs/PixelwiseRegression21Joints-p2Yj0yPh/lib/python3.10/site-packages/pyzed/sl.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZN2sl24SpatialMappingParametersC1ENS0_18MAPPING_RESOLUTIONENS0_13MAPPING_RANGEEibbbNS0_16SPATIAL_MAP_TYPEEi
```

### ZED Camera model

ZED Mini

### Environment

```shell
OS : Ubuntu 22.04.2 LTS
CPU : Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
GPU : Nvidia RTX 2070
ZED SDK : 4.0
```

rkcosner, 2023-06-29:
I'm running into the same issue with: 

Camera: Zed 2i 
OS: Ubuntu 22.04.2 LTS 
CPU: AMD® Ryzen 9 7950x 16-core processor × 32
GPU: Nvidia RTX 3090
ZED SDK:  4.0 
Python: 3.10

Uniqueml, 2023-06-30:
Same issue on Orin agx with Zed X camera when running any python3 example.

rkcosner, 2023-07-05:
I was running  ZED SDK 4.0.3. Upgrading to 4.0.5 fixed this issue

GabrielaHruba, 2023-08-01:
I have the same issue on ZED Box Nvidia Xavier NX, but only when I switch to root. When I'm logged in normal user, everything works fine. 

Camera: Zed 2i
ZED SDK: 4.0
Python 3.8
linux_aarch64

felixshing, 2023-08-19:
Any solution on this? I met the same error...

juniniho, 2023-09-04:
I met the same import error with ZED sdk 4.0.5 ,but I solved it by upgrade ZED sdk to 4.0.6.
the python tutorial works fine under the environment below
> OS : Windows11
Graphics: Nvidia RTX 2060
CUDA: 12.2
Python: 3.10.11
Zed SDK: 4.0.6
Camera: Zed 2i

pakshi10, 2023-10-03:
Goto where zed is installed
cd /usr/local/zed
There will be a file : get_python_api.py
RUN python3 get_python_api.py. This will install pyzed,numpy and cpython.

astringfield, 2023-10-03:
> I met the same import error with ZED sdk 4.0.5 ,but I solved it by upgrade ZED sdk to 4.0.6. the python tutorial works fine under the environment below
> 
> > OS : Windows11
> > Graphics: Nvidia RTX 2060
> > CUDA: 12.2
> > Python: 3.10.11
> > Zed SDK: 4.0.6
> > Camera: Zed 2i

@juniniho can I ask - what's the command for printing out the environment details as you've shown in your response? I'd love to be able to know the version number of my ZED SDK.

multicoremanticore, 2023-10-20:
The easiest way to get your zed SDK version is to run any of the tools in /usr/local/zed/tools
The SDK version will be printed in the lower left corner of them.

closed 2023-11-26 by github-actions[bot]

TomTheGoose, 2024-04-14:
Hey, I had the same problem and finally managed to solve it, leaving my solution here in case it helps anyone.

The issue comes from a mismatch between your ZED SDK version and your pyzed version. I initially thought mine where the same, but instead I had two pyzed versions in different directories! Steps to reproduce: 

1. Firstly run the zed diagnostic to confirm your SDK version.
2. Install the relevant pyzed module **take note of the install location**
3. Run your program again, when you get the error message, check the import error directory, if its not the same as the one shown on your install location, then this is your issue, your program is trying to use an outdated / corrupted version of pyzed.

If so, uninstall both pyzed versions (for me, it required the following):
`/home/tom/anaconda3/bin/python3.9 -m pip uninstall pyzed`
`python3.8 -m pip uninstall pyzed`

Then reinstall pyzed, specifying the correct directory