stereolabs/zed-sdk #589: Memory use increases when frequently open & close cameras

george-free, 2023-09-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I write a simple python script to grab images using zed-sdk, I found that If I close camera frequently, the memory increased.
script like this
```import pyzed.sl as sl
cam = sl.Camera()
while True:
    cam.open()
    cam.grab(settings)
    image = sl.Mat()
    depth = sl.Mat()
    cam.retrieve_image(image, sl.VIEW.LEFT)
    cam.retrieve_measure(depth, sl.MEASURE.DEPTH)
    cam.close()
```
The Images can be grabbed correctly, but the memory will continue to increase.
However, if get the `open` & `close` out of the while loop, the problem disappeared.
like,
```import pyzed.sl as sl
cam = sl.Camera()
cam.open()
while True:
    cam.grab(settings)
    image = sl.Mat()
    depth = sl.Mat()
    cam.retrieve_image(image, sl.VIEW.LEFT)
    cam.retrieve_measure(depth, sl.MEASURE.DEPTH)
cam.close()
```

### Steps to Reproduce

1. install sdk and python-sdk
2. grab images in python
3. run the program before 
...

### Expected Result

The memory will not increase in both the way as the camera is closed inside the loop.

### Actual Result

When I frequently open and close the camera, the system's memory increase, like a memory leak.

### ZED Camera model

ZED2i

### Environment

```shell
OS: ubuntu 20.04
python: 3.11.3
zed-sdk: 4.0.3
```

Myzhar, 2023-09-14:
Hi @george-free 
can you please test if the same problem is present with the latest SDK v4.0.6?
The new version brings fixes and improvements.

george-free, 2023-09-18:
Hi @Myzhar 
Thank you for your suggestion and I have tested it last weekend.
After I upgraded the zed SDK to v4.0.6, unfortunately, the issue remained.
The memory before running the script is
[image]
The memory after 66-hours running is
[image]

Finally, I found that the memory could not increase to `core dump` but kept at a high level(41.0%).

closed 2023-10-25 by github-actions[bot]