stereolabs/zed-sdk #590: ROI seems to have no effect on exposure digital-gain and analog-gain settings

DerekvdVen, 2023-09-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

For our application we want to set the sl.VIDEO_SETTINGS.AEC_AGC_ROI to only the bottom half of the image. While experimenting with setting the ROI I found that when setting it to a corner of the image, the exposure and gain still change a lot when darkening non ROI parts of the image. Only the digital gain seems to start running but that's just because the image becomes so dark. Either I'm doing something wrong, or the ROI seems to have no effect. 

Image without dark object (exp32999, analog gain: 1085, digital gain: 1:
[image]

Image with dark object not in ROI (exp32999, analog gain: 10516, digital gain: 1)
[image]

Image with dark object in entire image (exp32999, analog gain: 16000, digital gain: 111)
[image]

My code:
[take_images.txt](https://github.com/stereolabs/zed-sdk/files/12610178/take_images.txt)

### Steps to Reproduce

Take my code and run it. Hold a black object in front of the image and see the values change. 

### Expected Result

I would expect the exposure gain only to change when a significant lighting effect takes place inside the ROI. 

### Actual Result

The exposure and gain still change based on the entire image as roi. 

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu
CPU: arm64
GPU: Ampere GPU and a 6-core ARM CPU
ZED SDK version: 4.0.6
Other: It's not the ZED camera, it's the ZED X MINI (not in the list)
```

### Anything else?

It's not the ZED camera, it's the ZED X MINI (not in the list)

DerekvdVen, 2023-09-19:
I suspect it may have to do with our rotating the camera, am looking into this.

DerekvdVen, 2023-09-20:
I did make a mistake with the camera, but when setting the ROI correctly, the results still look a bit strange. It still responds more to bright parts outside the ROI than pixel values inside it.

P-yver, 2023-10-27:
Hi,
I will have a look at this but it may be due to the fact that the ROI is too small.

DerekvdVen, 2023-11-03:
Hi,

Thanks for taking a look. 

The code I sent you at first has a mistake in it, but after fixing we still face the situation same situation. 

We set the ROI to the bottom half of the image. Then, when we cover the top half with a black screen, we still see significant lighting changes in the bottom. Maybe this is expected because less light is entering the bottom half? But I would expect that the pixel values in the lower half don't really change and we would not get such a significant increase in lighting. 

[image]

closed 2023-12-10 by github-actions[bot]