stereolabs/zed-sdk #593: Can't run streaming_render.py

pfcouto, 2023-10-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am inside a distrobox container created using 

`
distrobox create --nvidia --name ubuntu2204-nvidia --image ubuntu:latest
`

I am able to perform `nvidia-smi` and get a positive result. However, ZED_Diagnostic says I don't have CUDA version 12.1, since my version is 12.2. How can I install cuda 12.1 inside a distrobox container? But this is not my main issue.

I am trying to run the script streaming_render.py using `python streaming_render.py` and I got the following error

```
Traceback (most recent call last):
  File "/home/pedrobolsa/Documents/Escola/Investigacao/DistroUbuntu2204/ZED/zed-sdk/camera streaming/sender/python/streaming_sender.py", line 94, in <module>
    main()
  File "/home/pedrobolsa/Documents/Escola/Investigacao/DistroUbuntu2204/ZED/zed-sdk/camera streaming/sender/python/streaming_sender.py", line 58, in main
    init.sdk_verbose = True
TypeError: Argument 'value' has incorrect type (expected int, got bool)
```

I wanted to try and fix this issue, but I can't see any documentation abot the SDK... 

### Steps to Reproduce

1. distrobox create --nvidia --name ubuntu2204-nvidia --image ubuntu:latest
2. distrobox enter ubuntu2204-nvidia
3. chmod +x ZED_SDK_Ubuntu22_cuda12.1_v4.0.7.zstd
4. ./ZED_SDK_Ubuntu22_cuda12.1_v4.0.7.zstd
5. python3 streaming_sender.py
...

### Expected Result

The script to run successfully

### Actual Result

The error:

```
Traceback (most recent call last):
  File "/home/pedrobolsa/Documents/Escola/Investigacao/DistroUbuntu2204/ZED/zed-sdk/camera streaming/sender/python/streaming_sender.py", line 94, in <module>
    main()
  File "/home/pedrobolsa/Documents/Escola/Investigacao/DistroUbuntu2204/ZED/zed-sdk/camera streaming/sender/python/streaming_sender.py", line 58, in main
    init.sdk_verbose = True
TypeError: Argument 'value' has incorrect type (expected int, got bool)

```

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
CPU: x86
ZED sdk version: 4.0 Early Version
Python version: Python 3.10.12
```

### Anything else?

I would appreciate, if I had not to create a new partition for Ubuntu. Currently, I am using Fedora, and will probably switch to Arch.

roalchaq, 2023-10-02:
I got the same error while running tutorial 1 (hello_zed.py). I see here: https://github.com/stereolabs/zed-python-api that there was a commit 3 months ago to fix this problem (29d625d). Probably, the installation script for the zed_python_api (get_python_api.py) is not installing the corrected version?

pfcouto, 2023-10-02:
I fixed it by changing from True to 1. I found the documentation and it says that this should be a integer, not boolean. 

About the python API, when I run the command `python get_python_api.py` this is my output. It does output some errors, but nothing related to the installation of the API I believe.

```
-> Downloading to '/usr/local/zed'
Detected platform: 
	 linux_x86_64
	 Python 3.10
	 ZED SDK 4.0
-> Checking if https://download.stereolabs.com/zedsdk/4.0/whl/linux_x86_64/pyzed-4.0-cp310-cp310-linux_x86_64.whl exists and is available
-> Found ! Downloading python package into /usr/local/zed/pyzed-4.0-cp310-cp310-linux_x86_64.whl
-> Installing necessary dependencies
Requirement already satisfied: numpy in /home/pedrobolsa/.local/lib/python3.10/site-packages (1.26.0)
Processing ./pyzed-4.0-cp310-cp310-linux_x86_64.whl
Collecting numpy>=1.13
  Using cached numpy-1.26.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (18.2 MB)
Collecting cython>=3.0.0
  Using cached Cython-3.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.6 MB)
Installing collected packages: numpy, cython, pyzed
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
gensim 4.3.0 requires FuzzyTM>=0.4.0, which is not installed.
numba 0.56.4 requires numpy<1.24,>=1.18, but you have numpy 1.26.0 which is incompatible.
darwin-py 0.8.41 requires pillow<10.0.0,>=9.4.0, but you have pillow 10.0.1 which is incompatible.
Successfully installed cython-3.0.2 numpy-1.26.0 pyzed-4.0
Done
  To install it later or on a different environment run : 
 python -m pip install --ignore-installed /usr/local/zed/pyzed-4.0-cp310-cp310-linux_x86_64.whl
```

By the way, do you know of any sample or tutorial that captures the RGB and Depth streams and shows them using cv2, or other option? Thanks!

closed 2023-11-07 by github-actions[bot]

PHChenGit, 2024-02-02:
I also face the same error
Are there any solutions?

sohyeonlm, 2024-03-08:
init_params.sdk_verbose = 1 then init_params.sdk_verbose = True
or
init_params.sdk_verbose = 0 then init_params.sdk_verbose = False

sohyeonlm, 2024-03-08:
> I also face the same error Are there any solutions?

init_params.sdk_verbose = 1 then init_params.sdk_verbose = True
or
init_params.sdk_verbose = 0 then init_params.sdk_verbose = False