stereolabs/zed-sdk #599: Body tracking is not working

zanna80, 2023-11-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I have installed each component that is required but when i start body_tracking.py I encounter this error: 

Running Body Tracking sample ... Press 'q' to quit, or 'm' to pause or restart
[Sample] Using default resolution
Traceback (most recent call last):
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\body_tracking.py", line 164, in <module>
    main() 
    ^^^^^^
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\body_tracking.py", line 119, in main
    viewer.init(camera_info.camera_configuration.calibration_parameters.left_cam, body_param.enable_tracking,body_param.body_format)
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\ogl_viewer\viewer.py", line 435, in init
    glutCreateWindow("ZED Body Tracking")
  File "C:\Users\aless\TestZed2\pyopengl\OpenGL\GLUT\special.py", line 73, in glutCreateWindow
    return __glutCreateWindowWithExit(title, _exitfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ctypes.ArgumentError: argument 1: TypeError: wrong type
PS C:\Users\aless\TestZed2\zed-sdk> 

### Steps to Reproduce

1. Just run the python file
2.
3.
...

### Expected Result

Open the camera and see the joints

### Actual Result

This error: 

Running Body Tracking sample ... Press 'q' to quit, or 'm' to pause or restart
[Sample] Using default resolution
Traceback (most recent call last):
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\body_tracking.py", line 164, in <module>
    main() 
    ^^^^^^
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\body_tracking.py", line 119, in main
    viewer.init(camera_info.camera_configuration.calibration_parameters.left_cam, body_param.enable_tracking,body_param.body_format)
  File "c:\Users\aless\TestZed2\zed-sdk\body tracking\body tracking\python\ogl_viewer\viewer.py", line 435, in init
    glutCreateWindow("ZED Body Tracking")
  File "C:\Users\aless\TestZed2\pyopengl\OpenGL\GLUT\special.py", line 73, in glutCreateWindow
    return __glutCreateWindowWithExit(title, _exitfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ctypes.ArgumentError: argument 1: TypeError: wrong type
PS C:\Users\aless\TestZed2\zed-sdk> 

### ZED Camera model

ZED2

### Environment

```shell
OS: Windows 11 Home
CPU: Intel Core i7
GPU: NVIDIA GEFORCE RTX 3050  
ZED SDK: 4.0.7
```

andrsngy, 2023-11-15:
I am struggling with exactly the same issue.
Have you found any solution yet?

zanna80, 2023-11-15:
Yes i find this solution:
https://codeyarns.com/tech/2012-04-26-pyopengl-glut-ctypes-error.html#gsc.tab=0
but i ecountered an other problem: body tracking screen remain black.
Please telll me if you find a way!

andrsngy, 2023-11-15:
Hey, I've made it! :)
I uninstalled all the python versions from the installed apps on the computer.
Than I deleted all the paths, followed this tutorial: https://www.pythoncentral.io/how-to-uninstall-python/
I installed python 64bit version and the corresponding PyOpenGL and PyOpenGL_accelerate from here: https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopengl
It is super important to choose the right version. You can check it in terminal by typing: `python --version`
Mine was: 3.10.6
I downloaded PyOpenGL_accelerate-3.1.6-cp310-cp310-win_amd64.whl 
                                                                    """"""" """""""-this number should match with python version
and PyOpenGL-3.1.6-cp310-cp310-win_amd64.whl
                                   """""" """""""-this number should match with python version
I installed it from terminal by: 
`pip install PyOpenGL-3.1.6-cp310-cp310-win_amd64.whl`
`pip install PyOpenGL_accelerate-3.1.6-cp310-cp310-win_amd64.whl`

And it worked!
I hope it will work for you too :)

zanna80, 2023-11-16:
Hi, can you please tell me the version of each software you have?

- ZED SKD
- CUDA
- PYTHON
- PYOPENGL

I'll try to replicate your configuration

andrsngy, 2023-11-25:
Hi,

ZED SDK: ZED_SDK_Windows_cuda12.1_v4.0.8.exe
CUDA: 12.1
PYTHON: 3.10.6 (64-bit)
PYOPENGL: PyOpenGL-3.1.6-cp310-cp310-win_amd64.whl, PyOpenGL_accelerate-3.1.6-cp310-cp310-win_amd64.whl

closed 2023-12-31 by github-actions[bot]