stereolabs/zed-sdk #601: Calibration tool doesn't work with ZED X MINI

gorghino, 2023-11-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi all, 
We recently bought 2x ZEDX Mini and we mounted on a ORIN NX powered drone.
We wanted to calibrate the cameras but apparently the tool doesn't work: the camera is detected and we can see the chessboard without any errors. The two aligners are not moving at all and no matter how we move the camera, the blue dot doesn't appear.

We tried with different monitors without success.

The camera is working, since the other tools (Diagnostics, Explorer, etc.) are working fine.
Only ONE camera is attached to the capture board. The capture board is then connected to the Orin NX via MIPI/CSI.

```
nvidia@nvidia-orin:~$ /usr/local/zed/tools/ZED_Calibration
Tracking worker initialized.
ZED 6 SN -1
Opening camera ...
  Detected FW  :  2001
  Detected S/N  :  *Removed*
ZED 5 SN *Removed*
```

[image]

https://github.com/stereolabs/zed-sdk/assets/1898283/b877a694-0c86-435d-998f-c5011932dbce

### Steps to Reproduce

1. Get and ORIN NX with Jetpack 5.1.1 and a ZED X MINI 
2. Power the system and the capture card
3. Run the ZED_Calibration provided with the SDK

### Expected Result

I expect to see the blue dot as the tutorial explains.

### Actual Result

The chessboard seems freezed and nothing is moving in the window.

### ZED Camera model

ZED Mini

### Environment

```shell
OS: NVIDIA Jetpack 5.1.1 - Ubuntu 20.04
CPU: ORIN NX
GPU: ORIN NX
ZED SDK Version: 4.0.8
```

### Anything else?

There is no ZED X MINI in the camera model form above.

gorghino, 2023-11-14:
I updated the issue above.
I tried the same procedure with a ZEDX (not mini) but the calibration tool freezes.

I then checked the dmesg with the ZEDX Mini and the log was full of nvargus warnings about "Wrong frequency range" similar to this [thread](https://forums.developer.nvidia.com/t/nvargus-sending-wrong-frequency-range-message/260353).

I then tried running nvargus-daemon with enableCamCoreLogs=5 and I collected the two logs that may help you with debugging.

[dmesg.log](https://github.com/stereolabs/zed-sdk/files/13348639/dmesg.log)
[zedX_argus.log](https://github.com/stereolabs/zed-sdk/files/13348640/zedX_argus.log)
[zedXMini_argus.log](https://github.com/stereolabs/zed-sdk/files/13348648/zedXMini_argus.log)

gorghino, 2023-12-15:
I reported the issue by email. They claimed it will be fixed in the next SDK version

gorghino, 2024-01-15:
Not stale

cblberlin, 2024-02-06:
> Not stale

It appears that this company is not addressing the bugs, as many of the issues are simply being left unresolved.

closed 2024-02-07 by Myzhar

Arnoldtj, 2024-09-04:
We had the same problem on our ZED Mini, frozen calibration screen at first.
So we downloaded the [default calibration file
](https://calib.stereolabs.com/) and placed in the correct folder.

But after we did that, funnily, the problem went away and we were able to see and move the blue dot, as the tutorial explains. So maybe it needs to detect a default file first in order to re-calibrate - a bit weird, but oh well.

Hope that helps!