stereolabs/zed-sdk #604: CMake of Download

bobdavies2000, 2023-12-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

CMake of the download fails - missing SL_Hub?  I am upgrading to the CUDA 12.1 with SL 4.08 Github repository (the latest available.)  I am running CMake-Gui.  Below is the log of what CMake reports.  This sounds like a simple error on my part but I am unable to see what I did wrong.  I upgraded to the latest CMake 3.28 in trying to fix this.

Bob Davies

Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
The C compiler identification is MSVC 19.36.32537.0
The CXX compiler identification is MSVC 19.36.32537.0
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - skipped
Detecting C compile features
Detecting C compile features - done
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - skipped
Detecting CXX compile features
Detecting CXX compile features - done
CMake Warning (dev) at body tracking/body tracking/cpp/CMakeLists.txt:14 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at body tracking/export/fbx export/cpp/CMakeLists.txt:16 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at body tracking/export/fbx export/cpp/CMakeLists.txt:33 (message):
  FBX library not found
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at body tracking/multi-camera/cpp/CMakeLists.txt:15 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v12.1 (found version "12.1") 
CMake Warning (dev) at camera control/cpp/CMakeLists.txt:16 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v12.1 (found suitable version "12.1", minimum required is "12") 
CMake Warning (dev) at camera streaming/receiver/cpp/CMakeLists.txt:16 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at camera streaming/sender/cpp/CMakeLists.txt:15 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

[... 235 lines cut ...]

### Expected Result

I was hoping to build all the samples.

### Actual Result

Nothing is able to be built if CMake is failing.

### ZED Camera model

ZED2i

### Environment

```shell
Windows 11 Pro build 22621
CUDA 12.1
Zed-SDK 4.08
CMake 3.28
```

### Anything else?

I was able to build the tutorials fine.

closed 2024-01-14 by github-actions[bot]

martinkorob, 2025-01-03:
Hey Bob any chance you resolved this? Seems like the Cmake list is old. I've tried editting it but can't seem to get it fully resolved.

bobdavies2000, 2025-01-03:
Martin - I just tested it again and it is still failing for me.  And that
is with CUDA 12.3 and the latest ZED SDK.  I would still like to get these
examples so please post if you find a solution.

Bob Davies

martinkorob, 2025-01-03:
I ended up using the file below because I really only wanted to run through the tutorials. Lemme know if that works for you.

[CMakeLists.txt](https://github.com/user-attachments/files/18301162/CMakeLists.txt)

martinkorob, 2025-01-03:
Note this is for C#