stereolabs/zed-sdk #605: Image Buffer Tearing

bobdavies2000, 2023-12-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

First - I like your cameras and have used them since they became available on Windows 10 and now Windows 11.

I was using CUDA 11.X and Zed-SDK 3.8 and everything was working fine.  I am moving to CUDA 12.1 and Zed-SDK 4.08.  My code was tearing images - color, right, point cloud - occasionally when I started using the new Zed 4.08.  I investigated and found that the images were tearing immediately after the zed.grab() in my C++ code.  

I figured a good test to share with you  would be tutorial 3.  I modified tutorial 3 to do an OpenCV imshow call with the color buffer (I added OpenCV to your tutorial 3 project.)  The tearing occurred just like in my code.  I wanted to get your full set of examples working but the CMake was failing (see previous Issue.)  Since I got the tutorials working I added 3 lines of code to your tutorial 3 so the main loop looks like this:

    while (i < 50) {
        // A new image is available if grab() returns ERROR_CODE::SUCCESS
        if (zed.grab() == ERROR_CODE::SUCCESS) {
            // Retrieve left image
            zed.retrieveImage(image, VIEW::LEFT);
            // here are my 3 lines of code.
            cv::Mat tmp = cv::Mat(image.getHeight(), image.getWidth(), CV_8UC4, image.getPtr<sl::uchar1>());
            cv::imshow("tmp", tmp);
            cv::waitKey(1);

The image successfully shows up most of the time but then tears.  Here is what the tear looks like:

[image]

Here is what it should look like:

[image]

### Steps to Reproduce

Use the modified Tutorial 3 to see the problem - see Description above.

### Expected Result

See the second image in the Description above.

### Actual Result

See the first image in the Description above.

### ZED Camera model

ZED2i

### Environment

```shell
Window 11
GPU is NVIDIA GeForce RTX 3060
CPU is Core i9-13900, 2000 MHz, 24 cores, 32 Logical cores
```

closed 2024-01-14 by github-actions[bot]

mpattnaik97, 2024-08-30:
I am facing a similar issue. I am using ZED SDK v4.1.3 with cuda v12.1 and cudnn v8.9.7. I have 6 Zed2i cameras each calling `grab()` in a separate thread. Then another thread calls the Fusion API function `Fusion::process()`. I get the images and detections for each camera and store these in 6 different buffers. I have 6 threads running for showing these images annotated with detections. Can someone help me with this, please?

amefabris, 2024-09-17:
same issue over here. I am using ZED SDK v4.0.8 with cuda v12.4. I have a Zed2i camera and I am just getting image through the zed_wrapper. Can someone help me with this, please?
I have a GPU - Nvidia RTX ADA 2000

bobdavies2000, 2024-09-17:
I had the same issue with tearing and I replaced the cable with a USB-C to
USB-C cable and the problem went away.

Bob Davies