stereolabs/zed-sdk #606: ZED Depth Viewer .exe does not open

zjj13253377486, 2023-12-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Now ZED Depth Viewer .exe does not open and flashes back. But before ZED Depth Viewer .exe was able to open and work normally. I ran the official ZED SDK samples”ZED SDK\samples\depth sensing\depth_sensing\python\depth_sensing.py”file with Python and got the error like "Running Depth Sensing sample ... Press 'Esc' to quit
Press 's' to save the point cloud
 - - - - - - - - - TEMPORARY_DSI - - - - - - - - - 
 Too much Memory already used !! 
 Trying get -14, only 10 left on 10 
 void __cdecl mYHMtzEBZ 
 Process finished with exit code -1073741819 (0xC0000005)”。 How do I fix this?

### Steps to Reproduce

Now ZED Depth Viewer .exe does not open and flashes back. But before ZED Depth Viewer .exe was able to open and work normally. I ran the official ZED SDK samples”ZED SDK\samples\depth sensing\depth_sensing\python\depth_sensing.py”file with Python and got the error like "Running Depth Sensing sample ... Press 'Esc' to quit
Press 's' to save the point cloud
 - - - - - - - - - TEMPORARY_DSI - - - - - - - - - 
 Too much Memory already used !! 
 Trying get -14, only 10 left on 10 
 void __cdecl mYHMtzEBZ 
 Process finished with exit code -1073741819 (0xC0000005)”。 How do I fix this?

### Expected Result

Open ZED Depth Viewer .exe

### Actual Result

"Running Depth Sensing sample ... Press 'Esc' to quit
Press 's' to save the point cloud
 - - - - - - - - - TEMPORARY_DSI - - - - - - - - - 
 Too much Memory already used !! 
 Trying get -14, only 10 left on 10 
 void __cdecl mYHMtzEBZ 
 Process finished with exit code -1073741819 (0xC0000005)”

### ZED Camera model

ZED2i

### Environment

```shell
OS:win11
CPU:13th Gen Intel(R) Core(TM)i9-13900H
GPU:NVIDIA GeForce RTX 4060
ZED SDK:4.0
```

braoutch, 2024-01-18:
Hi, can you send a report of ZED_Diagnostics ?
Also, did you try to reinstall the latest SDK (4.0.8) ?
Can you try to open ZED Depth Viewer from a console, and send here the output of the console ?

Myzhar, 2024-02-07:
Please use the official customer support channels: [support@stereolabs.com](mailto:support@stereolabs.com) or https://community.stereolabs.com/

GitHub issues should be used to report problems of the repository they refer to.

closed 2024-02-07 by Myzhar