stereolabs/zed-sdk #607: Jetson Orin NX - USB Camera not detected

saberkaygusuz, 2023-12-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

```
ZED Diagnostic Tool

Cannot find CUDNN Diagnostic : 35%
- Running ZED SDK Diagnostic : Warning
    Object detection module corrupted. Please reinstall the ZED SDK and enable the Object Detection option.
    ZED SDK version: 4.0.8
    CUDA version: V11.4.315
- Running Processor Diagnostic : OK  
    Processor:   ARMv8 Processor rev 1 (v8l)
    Motherboard:  NVIDIA orin_nx by PCB Arts, Unknown
Error: unable to open display stic : 50%
- Running Graphics Card Diagnostic : OK  
    Graphics card:  Orin
FileLocker - Thread 281472601921792 - UNLOCK - Already unlocked.
FileLocker - Thread 281472601921792 - UNLOCK - Already unlocked.
- Running Devices Diagnostic : Warning
    No GMSL driver found. USB camera are not impacted by this.
FileLocker - Thread 281472601921792 - UNLOCK - Already unlocked.
FileLocker - Thread 281472601921792 - UNLOCK - Already unlocked.
- Running Camera Test Diagnostic : Failed
    Camera not detected  Make sure the camera is plugged in or try another USB 3.0 port.

Diagnostic finished, a file has been generated : ZED_Diagnostic_Results.json
```

[ZED_Diagnostic_Results_201223.txt](https://github.com/stereolabs/zed-sdk/files/13727858/ZED_Diagnostic_Results_201223.txt)

We're having a own NVIDIA Jetson based hardware, which has an Orin NX 16GB included. Between the Jetson Orin and the Camera we're using a USB Hub "VL822-Q7" from ViaLabs, it is an USB 3.1 Hub, which delivers enough power per Port and the bandwidth is also verified. There is running a JetPack 35.3.1 on the Hardware, we've installed this package "ZED_SDK_Tegra_L4T35.3_v4.0.8.zstd.run" 

When plugging it into a USB Port (or rebooting it via the Python API), i got the following dmesg output: 

```
[527183.768818] usb 1-2.4.2: reset full-speed USB device number 11 using tegra-xusb
[527183.882919] hid-generic 0003:2B03:F881.000B: hiddev96,hidraw0: USB HID v1.11 Device [STEREOLABS ZED-2i HID INTERFACE] on usb-3610000.xhci-2.4.2/input0
[527184.226823] usb 2-1.4: USB disconnect, device number 14
[527184.288911] usb 1-2.4.2: USB disconnect, device number 11
[527185.024217] usb 1-2.4.2: new full-speed USB device number 12 using tegra-xusb
[527185.136551] usb 1-2.4.2: New USB device found, idVendor=2b03, idProduct=f881, bcdDevice= 3.09
[527185.136561] usb 1-2.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[527185.136565] usb 1-2.4.2: Product: ZED-2i HID INTERFACE
[527185.136568] usb 1-2.4.2: Manufacturer: STEREOLABS
[527185.136572] usb 1-2.4.2: SerialNumber: 31855993
[527185.144723] hid-generic 0003:2B03:F881.000C: hiddev96,hidraw0: USB HID v1.11 Device [STEREOLABS ZED-2i HID INTERFACE] on usb-3610000.xhci-2.4.2/input0
[527185.205087] usb 2-1.4: new SuperSpeed Gen 1 USB device number 15 using tegra-xusb
[527185.229175] usb 2-1.4: config 1 has an invalid descriptor of length 0, skipping remainder of the config
[527185.239256] usb 2-1.4: config 1 has 1 interface, different from the descriptor's value: 2
[527185.248095] usb 2-1.4: config 1 interface 0 altsetting 0 has 0 endpoint descriptors, different from the interface descriptor's value: 1
[527185.262940] usb 2-1.4: New USB device found, idVendor=2b03, idProduct=f880, bcdDevice= 1.00
[527185.262943] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[527185.262946] usb 2-1.4: Product: ZED 2i
[527185.262947] usb 2-1.4: Manufacturer: Technologies, Inc.
[527185.262949] usb 2-1.4: SerialNumber: OV0001
[527185.268204] uvcvideo: Found UVC 1.10 device ZED 2i (2b03:f880)
```

We got, as expected, the SuperSpeed USB for the ZED 2i Camera. 

Also the lsusb -t looks good: 

```
root@edgekit-desktop:/home/edgekit# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 10000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 10000M
        |__ Port 4: Dev 15, If 0, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 4: Dev 9, If 0, Class=Hub, Driver=hub/2p, 480M
            |__ Port 2: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 4, If 0, Class=, Driver=, 480M
```

In the lsusb view, however i didn't see the correct ZED names: 

```ZED_Diagnostic_Results_201223
Bus 002 Device 015: ID 2b03:f880  
Bus 002 Device 002: ID 2109:0822 VIA Labs, Inc. USB3.1 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2109:8822 VIA Labs, Inc. 
Bus 001 Device 012: ID 2b03:f881 VIA Labs, Inc.          USB2.0 Hub             
Bus 001 Device 009: ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

### Steps to Reproduce

1. Plug in the Camera

### Expected Result

Running the camera &` ZED_Diagnostic -c` without any issue. 

### Actual Result

<b>Camera not detected</b> <br/> Make sure the camera is plugged in or try another USB 3.0 port."

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04 focal
Module: NVIDIA Jetson Orin NX (16GB ram)
Release: 5.10.104-tegra
CUDA: 11.4.315
```

### Anything else?

When interacting with the Python API and i want to receive some data from the camera, i got the following output in the dmesg: 

```
[526519.935081] tegra-i2c 3190000.i2c: I2C transfer timed out
[526520.047077] tegra-i2c 31b0000.i2c: I2C transfer timed out
[526520.159076] tegra-i2c 31c0000.i2c: I2C transfer timed out
[526520.270816] tegra-i2c 31e0000.i2c: I2C transfer timed out
```

Myzhar, 2024-02-07:
Please use the official customer support channels: [support@stereolabs.com](mailto:support@stereolabs.com) or https://community.stereolabs.com/

GitHub issues should be used to report problems of the repository they refer to.

closed 2024-02-07 by Myzhar

7anmay, 2024-07-26:
Were you able to find a solution to this issue @saberkaygusuz ?