stereolabs/zed-sdk #641: get_python_api.py installs numpy v2 which give ValueError due to binary incompatibility

26atul05, 2024-06-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

While doing environment setup on Ubuntu, I followed the instructions in [How to Install ZED SDK on Linux](https://www.stereolabs.com/docs/installation/linux) 

During SDK installation it asked following  question -> "Do you want to install the Python API (recommended) [Y/n] ?"

And it executes get_python_api.py, which installs numpy v2
`Collecting numpy>=1.13 (from pyzed==4.1)
  Using cached numpy-2.0.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (60 kB)`

Which leads to following error while running simple hello_zed.py

### Steps to Reproduce

1. Follow the instructions in [How to Install ZED SDK on Linux](https://www.stereolabs.com/docs/installation/linux) 
2. During the installation, you might have to answer some questions on the installation of dependencies, tools and samples. Type y for yes and n for no and hit Enter.
3. type Y and hit Enter for "Do you want to install the Python API (recommended) [Y/n] ?
4. Observe the numpy version installed
 
...

### Expected Result

get_python_api.py should install compatible numpy version with latest Zed SDK

### Actual Result

get_python_api.py, installs latest numpy which is not compatible with latest Zed SDK

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 22.04.4 LTS
CPU: Intel i7-13800Hx20
GPU: Nvidia GeForce RTX 4060
ZED SDK version: 4.1.2
```

### Anything else?

Workaround was installing numpy v1.26.4 which was last release for v1

gagapaga, 2024-07-01:
Did you find a solution? I have the same error.

26atul05, 2024-07-01:
Yes, I have mentioned the workaround at the end of the bug.
Uninstall numpy v2.0.0 and install last supported version of v1 that is v1.26.4 using pip

RubTalha, 2024-07-08:
https://stackoverflow.com/questions/78650222/valueerror-numpy-dtype-size-changed-may-indicate-binary-incompatibility-expec

closed 2024-08-13 by github-actions[bot]