stereolabs/zed-sdk #646: ZED SDK Download Link Broken

alberthli, 2024-08-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The links to download ZED SDKs seem to be broken.

### Steps to Reproduce

Go to the [website](https://www.stereolabs.com/developers/release) and attempt to download any SDK. For instance, going to the link [https://download.stereolabs.com/zedsdk/4.1/cu121/ubuntu22](https://download.stereolabs.com/zedsdk/4.1/cu121/ubuntu22) gives an "Unable to Connect" error for me (I'm using FireFox).

### Expected Result

I should be able to download the SDK.

### Actual Result

I can't.

### ZED Camera model

ZED

### Environment

```shell
N/A
```

sumo43, 2024-08-06:
I'm not able to download it either. Please fix!

closed 2024-09-12 by github-actions[bot]

jerrytamchiho, 2024-11-25:
Still unable to download it... its November