stereolabs/zed-sdk #654: Plug / Unplug for ZED camera detection?

Jarradmorden, 2024-09-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi,

Does anyone have an issue where the ZED camera is not recognized until the camera is unplugged and plugged back in when you turn on the jetson nano 2gb, this is making it difficult to create daemons to run the camera as soon as the jetson nano starts, I also have to run the explorer sometimes first.. At first I thought it was perhaps the USB power management, but I'm not sure any ideas?

This seems to be recurring every time I start the Nano.

Thanks for any advice.

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

It to work as soon as you plug it in

### Actual Result

it dont unless replugged in

### ZED Camera model

ZED

### Environment

```shell
Ubuntu 20
Jetson Nano 2gb
```

Jarradmorden, 2024-11-02:
bump bump bump bump. . .

Jarradmorden, 2024-11-12:
bump bump BUMP

closed 2024-12-19 by github-actions[bot]

jbcpollak, 2025-03-20:
Is this with a Zed 2i (USB) or Zed X (GMSL)?

Does it happen on other computers as well or just the Jetson?

We see other hot-plugging related issues with Zed 2i USB cameras on desktop computers.