stereolabs/zed-sdk #655: OpenCV (4.10.0) throws an error when running body_tracking.py example

shruthiR-fauna, 2024-10-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm trying to run the body tracking example provided [here](https://github.com/stereolabs/zed-sdk/tree/master/body%20tracking/body%20tracking/python) and OpenCV throws this error

> shruthi11@ocean:~/code/zed-sdk/body tracking/body tracking/python$ python3 body_tracking.py 
Running Body Tracking sample ... Press 'q' to quit, or 'm' to pause or restart
[2024-10-04 17:43:48 UTC][ZED][INFO] Logging level INFO
[2024-10-04 17:43:48 UTC][ZED][INFO] Logging level INFO
[Sample] Using default resolution
[2024-10-04 17:43:48 UTC][ZED][INFO] Logging level INFO
[2024-10-04 17:43:48 UTC][ZED][INFO] [Init]  Depth mode: ULTRA
[2024-10-04 17:43:49 UTC][ZED][INFO] [Init]  Camera successfully opened.
[2024-10-04 17:43:49 UTC][ZED][INFO] [Init]  Camera FW version: 1523
[2024-10-04 17:43:49 UTC][ZED][INFO] [Init]  Video mode: HD1080@30
[2024-10-04 17:43:49 UTC][ZED][INFO] [Init]  Serial Number: S/N 28587480
Traceback (most recent call last):
  File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/body_tracking.py", line 164, in <module>
    main() 
  File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/body_tracking.py", line 135, in main
    cv_viewer.render_2D(image_left_ocv,image_scale, bodies.body_list, body_param.enable_tracking, body_param.body_format)
  File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/cv_viewer/tracking_viewer.py", line 56, in render_2D
    cv2.addWeighted(left_display, 0.9, overlay, 0.1, 0.0, left_display)
cv2.error: OpenCV(4.10.0) :-1: error: (-5:Bad argument) in function 'addWeighted'
> Overload resolution failed:
>  - src1 is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'src1'

> Segmentation fault (core dumped)

### Steps to Reproduce

1. Install ZED SDK
2. Connect ZED2 camera to computer
3. Clone https://github.com/stereolabs/zed-sdk
4. Follow install instructions in the README.md and install dependencies
5. Run body_tracking.py
...

### Expected Result

I'm expecting a window that renders the image from the ZEd camrea and overlays the skeleton on top of it. 

### Actual Result

```
 File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/body_tracking.py", line 164, in <module>
    main() 
  File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/body_tracking.py", line 135, in main
    cv_viewer.render_2D(image_left_ocv,image_scale, bodies.body_list, body_param.enable_tracking, body_param.body_format)
  File "/home/rshruthi11/code/zed-sdk/body tracking/body tracking/python/cv_viewer/tracking_viewer.py", line 56, in render_2D
    cv2.addWeighted(left_display, 0.9, overlay, 0.1, 0.0, left_display)
cv2.error: OpenCV(4.10.0) :-1: error: (-5:Bad argument) in function 'addWeighted'
> Overload resolution failed:
>  - src1 is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'src1'

```

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 22.04
CPU: 13th Gen Intel® Core™ i9-13900K × 32
GPU: NVIDIA GeForce RTX 4080
ZED SDK version: 4.2
Python version: 3.10
OpenCV version: 4.10
```

Myzhar, 2024-10-05:
@shruthiR-fauna there's a known issue with OpenCV v4.10.0.
Try to use an older version

shruthiR-fauna, 2024-10-07:
That resolved the issue - thanks!

closed 2024-10-07 by shruthiR-fauna