stereolabs/zed-sdk #665: [Windows-Python SDK] "ERROR: Unable to find pyzed installation folder" occurs when non-English characters are present in the execution path of get_python_api.py due to a Windows system encoding(cp949).

ssaru, 2024-12-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

- The system encoding of Windows with a Korean locale is cp949.
- Python programs executed on Windows follow the system encoding.
- The logic in `get_python_api.py` attempts to locate the installation path of pyzed using subprocess. However, an exception occurs because the system encoding is cp949, while the program tries to interpret the output as utf-8.
- The resulting error is simply displayed as `"ERROR: Unable to find pyzed installation folder"` which makes it difficult for users to identify the root cause.
- To resolve this issue, set `os.environ["PYTHONIOENCODING"] = "utf-8"` at the beginning of `get_python_api.py` and modify the `get_pyzed_directory` function to use `lines = subprocess.check_output(call_list, encoding="utf-8", errors="replace").splitlines()` .

### Steps to Reproduce

1. Install Windows 10 and set the locale to Korean.
2. Create a Windows account with a name in Korean and install the ZED SDK.
3. Execute get_python_api.py.

### Expected Result

The program terminates normally without any error messages.

### Actual Result

"ERROR: Unable to find pyzed installation folder"

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows10
CPU: 13th Gen Intel i9-13900H
GPU: RTX-4070
ZED SDK version: 4.2.2
```

adujardin, 2024-12-16:
Thank you for the report and the solution you posted. We'll patch this for the next release

adujardin, 2025-01-17:
Just to let you know this was part of the released ZED SDK 4.2.3

closed 2025-01-17 by adujardin