stereolabs/zed-sdk #680: ZED SDK ubuntu 24 not downloadable

altineller, 2025-03-20:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

The zed sdk for ubuntu 24.04 is not downloadable. It will return an amazon screen saying permission is denied.

Best regards.
-C

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

download file

### Actual Result

amazon permission denied

### ZED Camera model

ZED

### Environment

```shell
ubuntu 24.04
```

altineller, 2025-03-20:
not only that, the 4.2.4 and 4.2.5 is also not downloadable. it will forward to 5.0.

adujardin, 2025-03-21:
The Ubuntu 24.04 link has been fixed. Could you post the 4.2.4/4.2.5 links that redirect to 5.0?

altineller, 2025-03-23:
https://www.stereolabs.com/en-tr/developers/release#82af3640d775

this is the 4.2.5 link, but when you click on it it will go to 5.0 page.

Best regards,
Can

altineller, 2025-03-23:
Also, I was able to install 5.0 and use it, with the exception of python-api.
If installed normally on 24.04, pip is unable to install number of packages, and asks for --break-system-packages flag.
If installed on an anaconda environment, it will complain about version of numpy.

I fired up zedfu, and I see that this version is much nicer, so thank you for the hard work.

Best Regards,
Can

closed 2025-04-28 by github-actions[bot]