stereolabs/zed-sdk #684: GMSL Camera Not Detected Despite Being Recognized by ZED Explorer in SDK 5.0

mjshiggins, 2025-03-25:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The ZED SDK 5.0 fails to open a ZED-X Mini GMSL camera despite the camera being properly connected and detectable by ZED Explorer. I can also view the live stereo feed using the ZED Explorer program, but I cannot run the Depth Viewer. This occurs in both Python and C++ implementations, suggesting a core SDK issue rather than a language binding problem. This appears to be a regression in SDK 5.0 as the same hardware setup was working in previous versions.

### Steps to Reproduce

1. Hardware Setup:

- ZED-X Mini camera
- Jetson Xavier NX running Jetpack 5.1.1
- Connected via GMSL2 to ZED Link Mono, which is connected to the Xavier CAM0
- Camera visible at /dev/i2c-10

2. Software Setup:

- ZED SDK 5.0
- Tested on both Python and C++ implementations

3. Minimal Code Example (tutorial 1 / hello zed)

```
input_type = sl.InputType()
input_type.set_from_serial_number(57210108, sl.BUS_TYPE.GMSL)
init = sl.InitParameters(depth_mode=sl.DEPTH_MODE.NEURAL,
                        coordinate_units=sl.UNIT.METER,
                        coordinate_system=sl.COORDINATE_SYSTEM.RIGHT_HANDED_Y_UP,
                        input_t=input_type)
zed = sl.Camera()
status = zed.open(init)
```

4. Verification Steps

- Run ZED_Explorer --all to confirm camera is detected:

```
     ## Cam  0  ##
     Model :  "ZED-X Mini"
     S/N :  57210108
     State :  "Camera Available"
     Path :  /dev/i2c-10
     ID :  0
     Type :  "GMSL2"
```
- Attempt to open camera using SDK or Depth Viewer

### Expected Result

The camera should initialize successfully since:

- ZED Explorer detects it as "Camera Available"
- The camera is properly connected via GMSL2
- The stereo camera streams can be displayed with ZED Explorer
- The correct serial number and bus type are specified

### Actual Result

Camera fails to open with the following error sequence:
```
[2025-03-25 19:00:11 UTC][ZED][INFO] Logging level INFO
[2025-03-25 19:00:12 UTC][ZED][INFO] Using GMSL input... Switched to default resolution HD1200
[2025-03-25 19:00:13 UTC][ZED][INFO] [Init]  Depth mode: NEURAL LIGHT
Error CAMERA NOT DETECTED, exit program.
[2025-03-25 19:00:16 UTC][ZED][WARNING] CAMERA NOT DETECTED in sl::ERROR_CODE sl::Camera::open(sl::InitParameters)
[2025-03-25 19:00:16 UTC][ZED][ERROR] [ZED] sl::Camera::Open has not been called, no Camera instance running.
```

### ZED Camera model

ZED Mini

### Environment

```shell
OS: 5.10.104-tegra
Jetson: Xavier NX
Jetpack: 5.1.1
ZED Device:
 Model :  "ZED-X Mini"
 S/N :  57210108
 State :  "Camera Available"
 Path :  /dev/i2c-10
 ID :  0
 Type :  "GMSL2"
ZED SDK: 5.0
```

Myzhar, 2025-03-25:
Hi @mjshiggins 
Did you also upgrade the [ZED X driver to v1.3.0](https://www.stereolabs.com/developers/drivers#jetpack-5-65732a5b7ac5)?

mjshiggins, 2025-03-25:
Looks like that did it, apologies for the issue noise, and thank you for the quick response.

closed 2025-03-25 by mjshiggins

robinvetsch, 2025-05-06:
Hi @mjshiggins 

I have a similar issue but with a ZED X One camera. I use the Quad capture card with four Zed X stereo cameras and one ZED X One camera, I can see all the cameras with ZED Explorer and ZED Diagnostics doesn't cause any problems either.

When I try to read the live images from the four stereo cameras and the monocular camera with Python, I get the error ‘NO_CAMERA_DETECTED’ for the ZED X One camera. If I only address three stereo cameras with the Python API then it works. If I change the order of the cameras in the code (starting with the ZED X One), the ZED X One is recognised, but one stereo camera is not.

I am using a Jetson Orin AGX 64GB and have the latest ZED X driver and SDK 5.0 installed.

In Python I use sl.CameraOne() and sl.InitParametersOne() for the ZED X One

@Myzhar  do you possibly have a solution for this?

Myzhar, 2025-05-06:
Is the camera a ZED X One GS or 4K?

robinvetsch, 2025-05-06:
@Myzhar 

I have the ZED X One GS.

robinvetsch, 2025-05-19:
Hi @Myzhar

Do you already have a solution?

We would like to do some tests with a real machine next week, so it would be great to have a fix for the issue.

Should we try to install a latest release of the SDK 5.0.1 RC?

Many thanks for your support.