stereolabs/zed-sdk #687: ZED_Depth_Viewer not working

wzds2015, 2025-04-09:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Getting this error since today. Don't know how to fix.

~$ ZED_Depth_Viewer 
[2025-04-09 18:33:10 UTC][ZED][INFO] Logging level INFO
[2025-04-09 18:33:10 UTC][ZED][INFO] [Init]  Depth mode: NEURAL
[2025-04-09 18:33:11 UTC][ZED][INFO] [Init]  Camera successfully opened.
[2025-04-09 18:33:11 UTC][ZED][INFO] [Init]  Camera FW version: 1523
[2025-04-09 18:33:11 UTC][ZED][INFO] [Init]  Video mode: HD720@30
[2025-04-09 18:33:11 UTC][ZED][INFO] [Init]  Serial Number: S/N 11650965
in bool ImageHandler::initialize(sl::Mat&) : Err [999]: unknown error.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in bool ImageHandler::initialize(sl::Mat&) : Err [999]: unknown error.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in bool ImageHandler::initialize(sl::Mat&) : Err [400]: invalid resource handle.
in void OGLWidget::Deinit() : Err [400]: invalid resource handle.
in virtual void InputThreadModul::run() : Err [400]: invalid resource handle.

### Steps to Reproduce

1. DEZ_Depth_Viewer
2.
3.
...

### Expected Result

depth display

### Actual Result

no depth

### ZED Camera model

ZED

### Environment

```shell
1. cuda info:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Fri_Jan__6_16:45:21_PST_2023
Cuda compilation tools, release 12.0, V12.0.140
Build cuda_12.0.r12.0/compiler.32267302_0

2. graphics card: RTX4080, driver: 570.124.06
3. System: ubuntu 24.04
4. ZED sdk file: ZED SDK for Ubuntu 24 5.0 EA
```

maheshreddy24, 2025-04-17:
Hey were you able to get the depth map?

closed 2025-05-24 by github-actions[bot]

EricYang3721, 2025-06-24:
Found this tutorial that fixed my issue https://support.stereolabs.com/hc/en-us/articles/8422008229143-How-can-I-solve-ZED-SDK-OpenGL-issues-under-Ubuntu