stereolabs/zed-sdk #698: Trying to do object detection and output as mp4 file with svo

Hateley123, 2025-05-17:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

So I have an svo file that I am trying to do image detection for and convert it to an mp4 file. However, everytime i extract the image i get cv2.error: OpenCV(4.11.0) :-1: error: (-5:Bad argument) in function 'imwrite'
> Overload resolution failed:
>  - img is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'img'.

I thought it might be something on my end with my script, so I copied your code here: https://github.com/stereolabs/zed-sdk/tree/master/recording/export/svo/python but i get the exact same error. Have you ever come across this before?

### Steps to Reproduce

1. Record an SVo
2. use code https://github.com/stereolabs/zed-sdk/tree/master/recording/export/svo/python
3. execute
...

### Expected Result

It should create a .mp4 or .avi file from the .svo file.

### Actual Result

[2025-05-17 21:26:31 UTC][ZED][INFO] Logging level INFO
[2025-05-17 21:26:31 UTC][ZED][INFO] Logging level INFO
[2025-05-17 21:26:31 UTC][ZED][INFO] Logging level INFO
[2025-05-17 21:26:31 UTC][ZED][INFO] [Init]  Depth mode: PERFORMANCE
[2025-05-17 21:26:31 UTC][ZED][INFO] [Init]  Serial Number: S/N 21532660
Converting SVO... Use Ctrl-C to interrupt conversion.
Traceback (most recent call last):
  File "/root/camera_workspace/scripts/zed_svo_test.py", line 199, in <module>
    main(opt)
  File "/root/camera_workspace/scripts/zed_svo_test.py", line 148, in main
    cv2.imwrite(str(filename1), left_image.get_data())
cv2.error: OpenCV(4.11.0) :-1: error: (-5:Bad argument) in function 'imwrite'
> Overload resolution failed:
>  - img is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'img'

### ZED Camera model

ZED2

### Environment

```shell
os: Linux
GPU: Nvidia
```

JujuDel, 2025-05-19:
Hello @Hateley123, 

`left_image.get_data()` returns a numpy array (see https://github.com/stereolabs/zed-python-api/blob/master/src/pyzed/sl.pyx#L6097), while OpenCV seems to complain that "_img is not a numpy array, neither a scalar_"

Could you confirm which pyzed version you are using and check what is the type of the value returned by `left_image.get_data()`

We've seen similar issues with OpenCV 4.10, maybe you can also try to re-install OpenCV with a version prior to that?

Hateley123, 2025-05-19:
Thanks for letting me know. I downloaded open cv 4.9 and it works now. You can close the issue.

closed 2025-05-19 by Hateley123